StanfordVL/iGibson #1: collecting ... Segmentation fault

ChenhongyiYang, 2019-06-28:
Hi, 

I installed the GibsonV2 following the installation tutorial, and there is not problem during installation, but I cannot run the pytest or the examples in the example folder:

platform linux -- Python 3.6.8, pytest-4.3.1, py-1.8.0, pluggy-0.9.0
rootdir: /scratch2/gibsonV2/GibsonEnvV2, inifile:
plugins: remotedata-0.3.1, openfiles-0.3.2, doctestplus-0.3.0, arraydiff-0.3
collecting ... Segmentation fault

how can I fix this?
Thanks

fxia22, 2019-07-01:
can you try installing nvidia OpenGL drivers and make them visible?
`export LD_LIBRARY_PATH=/usr/lib/nvidia-<vvv>:$LD_LIBRARY_PATH`

GELIELEO, 2019-10-06:
> Hi,
>
> I installed the GibsonV2 following the installation tutorial, and there is not problem during installation, but I cannot run the pytest or the examples in the example folder:
>
> platform linux -- [...]
Hi, Have you solve your problem? please can you share some experience

GELIELEO, 2019-10-06:
Hi, Have you solve your problem? please can you share some experience

GELIELEO, 2019-10-06:
@ChenhongyiYang Hi, I meet the same problem as yours. Have you solve your problem? please can you share some experience?

ArbiterGe, 2020-09-17:
Me too，I don‘t install OpenGL drivers and Segmentation fault (core dumped)

roberto-martinmartin, 2021-06-30:
If you have this problem, please, try our latest version. It should solve the issue. If you are still encountering it, reopen the issue. Thanks!

closed 2021-06-30 by roberto-martinmartin