StanfordVL/iGibson #2: Unable to initialize EGL

krishanrana, 2019-07-23:
Hi, everything installed as requried however when running demo.py the above error is thrown.
Additionally I have noticed that there is no pytest file as shown in the README.

shsjxzh, 2019-07-28:
I also meet the similar problem. I successfully pass all the test but fail in running demo.py. The terminal information is as follows:
python demo.py 
/home/zxx5101/Desktop/GibsonEnvV2/gibson2/__init__.py:5: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  global_config = yaml.load(f)
using asset: path /home/zxx5101/Desktop/GibsonEnvV2/gibson2/assets dataset path: /home/zxx5101/Desktop/GibsonEnvV2/gibson2/assets/dataset
pybullet build time: Jul 26 2019 16:56:55
/home/zxx5101/Desktop/GibsonEnvV2/gibson2/utils/utils.py:8: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  config_data = yaml.load(f)
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1080 Ti/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 430.14
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 430.14
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 1080 Ti/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
number of devices found 2
Loaded EGL 1.5 after reload.
Failed to make context current (eglError: 12288)

ahmedosman, 2019-09-26:
@shsjxzh were you able to resolve this error?

shsjxzh, 2019-09-26:
@ahmedosman Sorry, no. I use another virtual environment called "ai2thor" as a substitution.

fxia22, 2019-09-26:
@ahmedosman Can you try to install with `USE_GLAD` set to `FALSE` in https://github.com/StanfordVL/GibsonEnvV2/blob/master/gibson2/core/render/CMakeLists.txt and try again? Thanks.

ahmedosman, 2019-09-29:
@fxia22 The error is resolved (Ubuntu 18.04). I think this issue could be closed.

GELIELEO, 2019-10-06:
@ahmedosman How did you solve your problem. Can you share your experience please?

ahmedosman, 2019-10-20:
> @ahmedosman Can you try to install with `USE_GLAD` set to `FALSE` in https://github.com/StanfordVL/GibsonEnvV2/blob/master/gibson2/core/render/CMakeLists.txt and try again? Thanks.

git pull to have the latest commit and follow the instructions suggest above.

chaundm, 2019-12-20:
> @ahmedosman Can you try to install with `USE_GLAD` set to `FALSE` in https://github.com/StanfordVL/GibsonEnvV2/blob/master/gibson2/core/render/CMakeLists.txt and try again? Thanks.

Hi Fei Xia,
Could you please let me know how to install this path?

chaundm, 2019-12-20:
I have the same problem with default setting (if I set USE_GLAD set to FALSE, I am unable to install gibson2). Here is my screenshot

[image]

lpeng8910, 2021-02-22:
Hello! I am also still running into the same error "Unable to initialize EGL" on Ubuntu 18.04. I followed steps 1-3 of the[ troubleshooting guide](http://svl.stanford.edu/igibson/docs/issues.html) and the output of my ```ldconfig -p | grep EGL``` is below:

```	
libEGL_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
```
 Tried setting ```USE_GLAD``` to ```FALSE``` but then ran into more build errors, i pasted the error part of the message below:
```
...
    [ 50%] Linking CXX executable /home/lisa/Documents/iGibson/gibson2/render/mesh_renderer/build/test_device
/usr/bin/ld: cannot find -lEGL
    collect2: error: ld returned 1 exit status
    CMakeFiles/query_devices.dir/build.make:148: recipe for target '/home/lisa/Documents/iGibson/gibson2/render/mesh_renderer/build/query_devices' failed
    make[2]: *** [/home/lisa/Documents/iGibson/gibson2/render/mesh_renderer/build/query_devices] Error 1
...
```
Any help would be appreciated.

mjlbach, 2021-07-24:
Hi, we've recently updated the installation guide: http://svl.stanford.edu/igibson/docs/installation.html#system-requirements

The most common reasons we have discovered that reproduce this EGL issue are:

1. There is a bug with cgroups on certain versions of the nvidia 465.x and 450.x drivers that prevents opening an EGL context on all but GPU 0, please update, or have your cluster administrator update your nvidia driver.
2. You are missing `libegl-dev` (ubuntu/debian package)

mjlbach, 2021-07-27:
Closing because I believe this was addressed with our latest documentation improvements. Please let us know if this is not the case.

closed 2021-07-27 by mjlbach