StanfordVL/iGibson #6: segment fault

GELIELEO, 2019-10-02:
I cd to the test dir and pytest. But the test does not pass.
I tried in both ubuntu16.04 and ubuntu18.04, neither worked.
The most possible problem is that both "querydevice" and "testdevice" can not work, but I can not solve. please help me.

when I run "./query_devices", it return "Segmentation fault (core dumped)"

fxia22, 2019-10-02:
Thanks for reporting, can you check the following:
1. is nvidia driver properly installed? You can check by running nvidia-smi
2. Are openGL libraries visible? You can do so by 
`export LD_LIBRARY_PATH=/usr/lib/nvidia-<vvv>:$LD_LIBRARY_PATH`
3. There are two ways of setting up openGL library, if the current installation doesn't work, you can try to install with USE_GLAD set to FALSE in https://github.com/StanfordVL/GibsonEnvV2/blob/master/gibson2/core/render/CMakeLists.txt

GELIELEO, 2019-10-03:
Thanks for your reply. I have tried all the methods you give but they do not work. my nividia driver is quite OK after nvidia-smi; And setting USE_GLAD to FALSE does not work too. There is one thing worth paying attention maybe, that is, I only can find dir 'nvidia' under the path /usr/lib/ while not 'nvidia-<vvv>'. I forget to install something maybe???

GELIELEO, 2019-10-03:
> Thanks for reporting, can you check the following:
>
> 1. is nvidia driver properly installed? You can check by running nvidia-smi
> 2. Are openGL libraries visible? You can do so by
>    `export [...]

GELIELEO, 2019-10-03:
I use valgrind to debug as well, and it tell me:

> ==24322== HEAP SUMMARY:
==24322==     in use at exit: 41,959 bytes in 165 blocks
==24322==   total heap usage: 212 allocs, 47 frees, 187,922 bytes allocated
==24322== 
==24322== Searching for pointers to 165 not-freed blocks
==24322== Checked 420,024 bytes
==24322== 
==24322== LEAK SUMMARY:
==24322==    definitely lost: 56 bytes in 1 blocks
==24322==    indirectly lost: 0 bytes in 0 blocks
==24322==      possibly lost: 80 bytes in 1 blocks
==24322==    still reachable: 41,823 bytes in 163 blocks
==24322==         suppressed: 0 bytes in 0 blocks
==24322== Rerun with --leak-check=full to see details of leaked memory
==24322== 
==24322== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
==24322== 
==24322== 1 errors in context 1 of 1:
==24322== Jump to the invalid address stated on the next line
==24322==    at 0x0: ???
==24322==    by 0x118BDF: main (in /home/ibasaw/causal_ws/GibsonEnvV2/gibson2/core/render/mesh_renderer/build/query_devices)
==24322==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==24322== 
==24322== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

fxia22, 2019-10-03:
Hi @GELIELEO, 

The egl program is based on https://github.com/erwincoumans/egl_example, can you also show the output of running ./egl_example after compiling that repo?

Also can you show the results of `ldconfig -p  | grep GL`?

GELIELEO, 2019-10-04:
It also shows me segment fault after I try  https://github.com/erwincoumans/egl_example.

And the result of ldconfig -p | grep GL is listed as folllow:
> libvtkRenderingVolumeOpenGLTCL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGLTCL-6.3.so.6.3
	libvtkRenderingVolumeOpenGLTCL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGLTCL-6.3.so
	libvtkRenderingVolumeOpenGLPython27D-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGLPython27D-6.3.so.6.3
	libvtkRenderingVolumeOpenGLPython27D-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGLPython27D-6.3.so
	libvtkRenderingVolumeOpenGL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so.6.3
	libvtkRenderingVolumeOpenGL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so
	libvtkRenderingOpenGLTCL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGLTCL-6.3.so.6.3
	libvtkRenderingOpenGLTCL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGLTCL-6.3.so
	libvtkRenderingOpenGLPython27D-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGLPython27D-6.3.so.6.3
	libvtkRenderingOpenGLPython27D-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGLPython27D-6.3.so
	libvtkRenderingOpenGL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3
	libvtkRenderingOpenGL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so
	libvtkRenderingGL2PSPython27D-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGL2PSPython27D-6.3.so.6.3
	libvtkRenderingGL2PSPython27D-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGL2PSPython27D-6.3.so
	libvtkRenderingGL2PS-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGL2PS-6.3.so.6.3
	libvtkRenderingGL2PS-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGL2PS-6.3.so
	libvtkRenderingGLtoPSTCL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGLtoPSTCL-6.3.so.6.3
	libvtkRenderingGLtoPSTCL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingGLtoPSTCL-6.3.so
	libvtkRenderingContextOpenGLTCL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGLTCL-6.3.so.6.3
	libvtkRenderingContextOpenGLTCL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGLTCL-6.3.so
	libvtkRenderingContextOpenGLPython27D-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGLPython27D-6.3.so.6.3
	libvtkRenderingContextOpenGLPython27D-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGLPython27D-6.3.so
	libvtkRenderingContextOpenGL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGL-6.3.so.6.3
	libvtkRenderingContextOpenGL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkRenderingContextOpenGL-6.3.so
	libvtkGUISupportQtOpenGL-6.3.so.6.3 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQtOpenGL-6.3.so.6.3
	libvtkGUISupportQtOpenGL-6.3.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQtOpenGL-6.3.so
	libQt5OpenGL.so.5 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5
	libQt5OpenGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so
	libOpenGL.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libOpenGL.so.0
	libOpenGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libOpenGL.so
	libGLdispatch.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0
	libGLdispatch.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLdispatch.so
	libGLX_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0
	libGLX.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLX.so.0
	libGLX.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLX.so
	libGLU.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLU.so.1
	libGLU.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLU.so
	libGLEW.so.2.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLEW.so.2.0
	libGLEW.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLEW.so
	libGLESv2.so.2 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLESv2.so.2
	libGLESv2.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLESv2.so
	libGLESv1_CM.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLESv1_CM.so.1
	libGLESv1_CM.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGLESv1_CM.so
	libGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGL.so.1
	libGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libGL.so
	libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
	libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
	libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so

GELIELEO, 2019-10-07:
I solve my problem after I do s software update. HAHAHAHAHA...

zubair-irshad, 2019-12-26:
@GELIELEO I am having the same problem. Can I ask how did you resolve this specifically?

closed 2020-05-20 by fxia22