StanfordVL/iGibson #17: Exception: Failed to reset robot without collision

chaundm, 2019-12-23:
Hi Fei XIa,

I have a trouble with running demo.py file. Most of things are work very well except for num_indices = 36. See my log file as followings: 
Could you please help to have a look at my problem?

using asset: path /home/cndm/Documents/GibsonEnvV2/gibson2/assets dataset path: /home/cndm/Documents/GibsonEnvV2/gibson2/assets/dataset
device 0 is available for rendering
device 1 is not available for rendering
device 2 is not available for rendering
device 3 is not available for rendering
using device 0
fisheye False
building scene: Ohoopee
floors [-2.3225653171539307, 0.03626394644379616, 2.945486545562744]
attrib.vertices =  699732
attrib.normals =  699732
attrib.texcoords =  625656
Num materials:  1
material_0
[1.0, 1.0, 1.0]
Num shapes:  1
{0: Material(type: texture, texture_id: 6, color: [0.5, 0.5, 0.5])}
(233244, 3) (233244, 3) (312828, 2)

material_id = 0
num_indices = 1373199
mesh_materials [0]
turtlebot/turtlebot.urdf
/home/cndm/Documents/GibsonEnvV2/gibson2/assets/models/turtlebot/kobuki_description/meshes/main_body.obj (0.0010000000474974513, 0.0, 0.051989998668432236) (0.0, 0.0, 0.0, 1.0) (1.0, 1.0, 1.0, 1.0) (1.0, 1.0, 1.0)
attrib.vertices =  12666
attrib.normals =  12666
attrib.texcoords =  9128
Num materials:  1
material_0
[1.0, 1.0, 1.0]
Num shapes:  1
{0: Material(type: texture, texture_id: 6, color: [0.5, 0.5, 0.5]), 1: Material(type: texture, texture_id: 7, color: [0.5, 0.5, 0.5]), 2: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0))}
(4222, 3) (4222, 3) (4564, 2)

material_id = 0
num_indices = 25320
mesh_materials [0, 1]
/home/cndm/Documents/GibsonEnvV2/gibson2/assets/models/mjcf_primitives/cube.obj (0.02, 0.01, 0.0) (0.0010000000474974513, 0.0, 0.051989998668432236) (0.0, 0.0, 0.0, 1.0) (1.0, 1.0, 1.0, 1.0)
attrib.vertices =  24
attrib.normals =  18
attrib.texcoords =  8
Num materials:  0
Num shapes:  1
{0: Material(type: texture, texture_id: 6, color: [0.5, 0.5, 0.5]), 1: Material(type: texture, texture_id: 7, color: [0.5, 0.5, 0.5]), 2: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0)), 3: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0))}
(8, 3) (6, 3) (4, 2)
cube
material_id = -1
num_indices = 36
mesh_materials [0, 1, 3]
/home/cndm/Documents/GibsonEnvV2/gibson2/assets/models/turtlebot/kobuki_description/meshes/wheel.obj (0.0, 0.0, 0.0) (0.0, 0.0, 0.0, 1.0) (1.0, 1.0, 1.0, 1.0) (1.0, 1.0, 1.0)
attrib.vertices =  7188
attrib.normals =  7188
attrib.texcoords =  5376
Num materials:  1
material_0
[1.0, 1.0, 1.0]
Num shapes:  1
{0: Material(type: texture, texture_id: 6, color: [0.5, 0.5, 0.5]), 1: Material(type: texture, texture_id: 7, color: [0.5, 0.5, 0.5]), 2: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0)), 3: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0)), 4: Material(type: texture, texture_id: 8, color: [0.5, 0.5, 0.5]), 5: Material(type: color, texture_id: None, color: (1.0, 1.0, 1.0))}
(2396, 3) (2396, 3) (2688, 2)

material_id = 0
num_indices = 14364
mesh_materials [0, 1, 3, 4]
/home/cndm/Documents/GibsonEnvV2/gibson2/assets/models/mjcf_primitives/cube.obj (0.0176, 0.017, 0.0) (0.0, 0.0, 0.0) (0.0, 0.0, 0.0, 1.0) (0.28125, 0.51953125, 0.92578125, 1.0)
attrib.vertices =  24
[... 175 lines cut ...]
    nav_env.reset()
  File "/home/cndm/Documents/GibsonEnvV2/gibson2/envs/locomotor_env.py", line 587, in reset
    state = super(NavigateRandomEnv, self).reset()
  File "/home/cndm/Documents/GibsonEnvV2/gibson2/envs/locomotor_env.py", line 519, in reset
    self.reset_agent()
  File "/home/cndm/Documents/GibsonEnvV2/gibson2/envs/locomotor_env.py", line 502, in reset_agent
    raise Exception("Failed to reset robot without collision")
Exception: Failed to reset robot without collision
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

shantanusingh16, 2019-12-26:
A fix for the same has been merged a few hours ago. Can you pull the latest code and try again?

fxia22, 2019-12-26:
It is caused by floor height being too high so every initialization ends up in collision. Please try the latest commit from master branch. Also, increasing `random_init_z_offset` in yaml file (to >0.15) can also solve the issue.

closed 2019-12-27 by chaundm