StanfordVL/iGibson #18: Super slow simulation after adding "rgb_filled' in output

zubair-irshad, 2019-12-23:
Simulations gets super slow after I add "rgb_filled" to my output within the config file. Any leads would be appreciated!

fxia22, 2019-12-26:
Currently the navigation environment is calculating the shortest path every frame so that bottlenecks the performance, I will make that optional.

fxia22, 2019-12-26:
By the way, are you using headless mode? Running the gui can significantly slow down the simulation.

zubair-irshad, 2019-12-26:
Thanks @fxia22 
Update: 
on CPU, fps significantly improved after optimizing frame skip and time step values
on GPU, I am getting ' Unable to initialize EGL' whereas EGL initializes when I set gpu_idx =0

I have tried the fixes mentioned here: https://github.com/StanfordVL/GibsonEnv/issues/106 (egl_examples also run smoothly) I would appreciate any leads regarding the EGL issue.

My system specs: Ubuntu 16, Nvidia- gtx 1080, Nvidia-440 driver, CUDA 9.0

zubair-irshad, 2019-12-27:
Thanks @fxia22
Update:
on GPU, fps value improves to around 30 after the following settings: 
Frame skip = 10, 
use_filter=false 
Mode= Headless 
display_ui = false

It seems still very low compared to the fps mentioned on the website. Would you recommend any further optimizations? 

My system specs: Ubuntu 16, Nvidia- gtx 1080, Nvidia-440 driver, CUDA 9.0

closed 2020-05-07 by zubair-irshad