StanfordVL/iGibson #19: Unable to run mesh_renderer_gpu_example.py (or anthing related to mesh render in examples directory)

chaundm, 2019-12-27:
I have run demo.py but I have a trouble when running file mesh_renderer_gpu_example.py caused by an error in the following log. Could you please have a look at it? I have tried to run other files related to mesh but having the same error

The log in my case is:

(py3-gibson) cndm@pi:/media/cndm/New Volume/Project-2/GibsonEnvV2/examples/demo$ python mesh_renderer_gpu_example.py 
using asset: path /media/cndm/New Volume/Project-2/GibsonEnvV2/gibson2/assets dataset path: /media/cndm/New Volume/Project-2/GibsonEnvV2/gibson2/assets/dataset
pybullet build time: Dec 17 2019 12:16:21
Traceback (most recent call last):
  File "mesh_renderer_gpu_example.py", line 10, in <module>
    model_path = sys.argv[1]
IndexError: list index out of range

fxia22, 2019-12-27:
You need to give it an obj file, for example you can render the mesh in the dataset you downloaded:
`python mesh_renderer_gpu_example.py GibsonEnvV2/gibson2/assets/dataset/Norvelt/mesh_z_up.obj`

chaundm, 2019-12-27:
Thank you Fei Xia,

I have tried it in that way and the problem is nearly solved because it raises another issue as followings:

[image]

Could you please help me to figure out where the issue come from?

fxia22, 2019-12-27:
You need to use your `GibsonEnvV2` directory, which is `/media/cndm/...`

chaundm, 2019-12-27:
It works properly, thank you so much

closed 2019-12-27 by chaundm