StanfordVL/iGibson #20: Example file for navigating like Gibson version 1?

chaundm, 2019-12-27:
Hi Fei Xia,

I would like to find a file in Gibson version 2, such as play_husky_camera.py like Gibson version 1. Do you have any file like that to navigate the agent?

Berk035, 2020-05-16:
I wonder it too. Thanks.

roberto-martinmartin, 2022-01-07:
This is an old message but in case someone looks for it in the future, you can look in the branch https://github.com/StanfordVL/iGibson/tree/cleanup-robot-demos and run the example robot_control_example.py. Selecting the husky robot with teleoperation and differential drive control leads to the same behavior as in the play_husky_camera.py of the Gibson paper. It should be easy to extract those options into its own application bypassing the selection procedure

roberto-martinmartin, 2022-01-07:
this example will be moved into master in the next days

closed 2022-01-07 by roberto-martinmartin