StanfordVL/iGibson #24: Ohopee scene does not render when using examples/demo/mesh_renderer_example.py

pirobot, 2020-01-21:
When I run the command:

python mesh_renderer_gpu_example.py ~/src/gibson-master/gibson2/assets/dataset/Ohopee

I see a stream of "Render: complete" messages but when the Figure 1 window appears, all I see is a black panel like this:

[image]

pirobot, 2020-01-21:
Perhaps this is relevant.  Near the top of the message output I see the lines:

using asset: path /home/patrick/src/gibson-master/gibson2/assets dataset path: /home/patrick/src/gibson-master/gibson2/assets/dataset
pybullet build time: Jan 20 2020 16:10:50
device 0 is available for rendering
using device 0
fisheye False
**Unable to load numpy_formathandler accelerator from OpenGL_accelerate**
Using cuda device 0

Note the line in bold.  I have verified that I have pyopengl==3.1.0 in my Gibson python3.6 conda environment.

fxia22, 2021-09-22:
This issue is outdated since we no longer use pyopengl. We invite you to use latest version of iGibson.

closed 2021-09-22 by fxia22