StanfordVL/iGibson #27: Space7 dataset for Gibson 2

MADHAVAN001, 2020-03-10:
Where can I find the space7 dataset for Gibson Env 2?

Some attributes like floors seem to be missing from V1

yj-Tang, 2021-06-14:
Hi, Have you found the space7 dataset? I also get stuck in the same problem.

roberto-martinmartin, 2021-06-30:
If you fill out the [form](https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform) and click submit, you should be able to download 2d3ds_for_igibson.zip. Let us know if this works for you.

closed 2021-06-30 by roberto-martinmartin

ChengshuLi, 2021-09-17:
@MADHAVAN001 

For Stanford 2d3ds and Matterport3D dataset, we have updated our dataset and documentation. Could you follow this new documentation and try again? Thanks!
http://svl.stanford.edu/igibson/docs/dataset.html