StanfordVL/iGibson #28: Build failing on mac

dhruvbatra, 2020-04-07:
Probably because it needs CUDA? 

```bash
dbatra at dbatra-mbp in ~/iGibson on master
$ pip install -e .
Obtaining file:///Users/dbatra/iGibson
Requirement already satisfied: numpy==1.16.6 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (1.16.6)
Requirement already satisfied: scipy==1.2.3 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (1.2.3)
Requirement already satisfied: gym==0.12 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (0.12.0)
Requirement already satisfied: PyYAML in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (5.3.1)
Requirement already satisfied: pybullet==2.6.4 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (2.6.4)
Requirement already satisfied: transforms3d==0.3.1 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (0.3.1)
Requirement already satisfied: tqdm==4.19.9 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (4.19.9)
Requirement already satisfied: Pillow==6.1.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (6.1.0)
Requirement already satisfied: matplotlib==2.2.5 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (2.2.5)
Requirement already satisfied: cloudpickle==1.2.2 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (1.2.2)
Requirement already satisfied: opencv-python==4.2.0.32 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (4.2.0.32)
Requirement already satisfied: aenum==2.2.3 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (2.2.3)
Requirement already satisfied: GPUtil==1.4.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (1.4.0)
Requirement already satisfied: networkx==2.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (2.0)
Requirement already satisfied: ipython==5.9.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (5.9.0)
Requirement already satisfied: pytest==4.6.9 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (4.6.9)
Requirement already satisfied: future in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gibson2==0.0.4) (0.18.2)
Requirement already satisfied: six in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gym==0.12->gibson2==0.0.4) (1.14.0)
Requirement already satisfied: pyglet>=1.2.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gym==0.12->gibson2==0.0.4) (1.5.3)
Requirement already satisfied: requests>=2.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from gym==0.12->gibson2==0.0.4) (2.23.0)
Requirement already satisfied: pytz in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from matplotlib==2.2.5->gibson2==0.0.4) (2019.3)
Requirement already satisfied: cycler>=0.10 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from matplotlib==2.2.5->gibson2==0.0.4) (0.10.0)
Requirement already satisfied: python-dateutil>=2.1 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from matplotlib==2.2.5->gibson2==0.0.4) (2.8.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from matplotlib==2.2.5->gibson2==0.0.4) (2.4.7)
Requirement already satisfied: kiwisolver>=1.0.1 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from matplotlib==2.2.5->gibson2==0.0.4) (1.2.0)
Requirement already satisfied: decorator>=4.1.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from networkx==2.0->gibson2==0.0.4) (4.4.2)
Requirement already satisfied: prompt-toolkit<2.0.0,>=1.0.4 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (1.0.18)
Requirement already satisfied: pickleshare in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (0.7.5)
Requirement already satisfied: simplegeneric>0.8 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (0.8.1)
Requirement already satisfied: appnope; sys_platform == "darwin" in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (0.1.0)
Requirement already satisfied: setuptools>=18.5 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (46.1.3.post20200330)
Requirement already satisfied: pygments in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (2.6.1)
Requirement already satisfied: traitlets>=4.2 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (4.3.3)
Requirement already satisfied: pexpect; sys_platform != "win32" in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from ipython==5.9.0->gibson2==0.0.4) (4.8.0)
Requirement already satisfied: attrs>=17.4.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (19.3.0)
Requirement already satisfied: more-itertools>=4.0.0; python_version > "2.7" in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (8.2.0)
Requirement already satisfied: packaging in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (20.3)
Requirement already satisfied: pluggy<1.0,>=0.12 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (0.13.1)
Requirement already satisfied: wcwidth in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (0.1.9)
Requirement already satisfied: py>=1.5.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (1.8.1)
Requirement already satisfied: importlib-metadata>=0.12; python_version < "3.8" in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (1.6.0)
Requirement already satisfied: atomicwrites>=1.0 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pytest==4.6.9->gibson2==0.0.4) (1.3.0)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from requests>=2.0->gym==0.12->gibson2==0.0.4) (1.25.8)
Requirement already satisfied: idna<3,>=2.5 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from requests>=2.0->gym==0.12->gibson2==0.0.4) (2.9)
Requirement already satisfied: chardet<4,>=3.0.2 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from requests>=2.0->gym==0.12->gibson2==0.0.4) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from requests>=2.0->gym==0.12->gibson2==0.0.4) (2020.4.5.1)
Requirement already satisfied: ipython-genutils in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from traitlets>=4.2->ipython==5.9.0->gibson2==0.0.4) (0.2.0)
Requirement already satisfied: ptyprocess>=0.5 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from pexpect; sys_platform != "win32"->ipython==5.9.0->gibson2==0.0.4) (0.6.0)
Requirement already satisfied: zipp>=0.5 in /anaconda3/envs/gibson2/lib/python3.6/site-packages (from importlib-metadata>=0.12; python_version < "3.8"->pytest==4.6.9->gibson2==0.0.4) (3.1.0)
Installing collected packages: gibson2
  Running setup.py develop for gibson2
    ERROR: Command errored out with exit status 1:
     command: /anaconda3/envs/gibson2/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/Users/dbatra/iGibson/setup.py'"'"'; __file__='"'"'/Users/dbatra/iGibson/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /Users/dbatra/iGibson/
    Complete output (60 lines):
    running develop
    running egg_info
    writing gibson2.egg-info/PKG-INFO
    writing dependency_links to gibson2.egg-info/dependency_links.txt
    writing requirements to gibson2.egg-info/requires.txt
    writing top-level names to gibson2.egg-info/top_level.txt
    reading manifest file 'gibson2.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'gibson2.egg-info/SOURCES.txt'
[... 28 lines cut ...]
        return distutils.core.setup(**attrs)
[... cut ...] tus 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /anaconda3/envs/gibson2/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/Users/dbatra/iGibson/setup.py'"'"'; __file__='"'"'/Users/dbatra/iGibson/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.
```

fxia22, 2020-04-13:
Hi Dhruv, 
Thanks for the question, building on Mac should be possible, with a few features disabled (for example render to tensor, and also need glfw3 context as OS X has pool EGL support). We are working on a platform-dependent CMakeLists to make building on mac easier.

dhruvbatra, 2020-04-13:
Makes sense. Good luck.

fxia22, 2020-06-15:
This feature is added in version 0.0.5 per @dhruvbatra 's suggestion. Thanks a lot!

closed 2020-06-15 by fxia22