StanfordVL/iGibson #30: Why don't I see Fetch's arm in RobotView?

Caselles, 2020-04-12:
Hello, 

I was wondering if in the RobotView window, I'm able to see Fetch's arm when it is in the vision field of the robot?

I don't see it, but I might be misusing the software so if someone from the development team could tell me if this feature is supported or not, that would be very helpful. Thanks !

fxia22, 2020-04-13:
Thanks for your question. By default, we don't render the robot body itself as some robots might have parts occluding its camera. It is written here:https://github.com/StanfordVL/iGibson/blob/master/gibson2/core/render/mesh_renderer/mesh_renderer_cpu.py#L757

If you want to render the robot body, you can simply change it from `self.render(modes=modes, hidden=[instance])`to `self.render(modes=modes)` We should have exposed this as a parameter passed to the renderer. We will update this in future releases.

Caselles, 2020-04-13:
Thank you so much, it works as intended. 

Closing this issue.

closed 2020-04-13 by Caselles