StanfordVL/iGibson #31: Difference between "trav_map" and "obstacle_map" in core/physics/scene.py

WSPeng, 2020-04-13:
Hi,
I see their source image are different but I don't understand the difference between "trav_map" and "obstacle_map". Could you explain their difference? 

Many thanks!

ChengshuLi, 2020-04-13:
Thanks for the question.

We use two different heuristics to generate trav_map and obstacle_map (e.g. floor_trav_0.png and floor_0.png). They are used together in the code to best represent the traversability of a scene.
https://github.com/StanfordVL/iGibson/blob/822ff1a77f8ee40dc72e14439ef8af1c92a48d38/gibson2/core/physics/scene.py#L218
Essentially, we take a conservative approach and treat an area as traversable or empty only if both trav_map and obstacle_map represent that area as empty (with pixel value 255)

For example, this is the "trav_map" of a scene called "Rs".
[image]

And this is the "obstacle_map" of the same scene.
[image]

WSPeng, 2020-04-13:
Thank you, I got it now.

WSPeng, 2020-04-14:
Hi, just go another related question, some image has name ended with "_V3". What is the difference between the "v3" map and normal map? Thank you!

ChengshuLi, 2020-04-22:
Sorry for my belated reply. I think "v3" map is generated using a slightly different heuristic. Neither of them are completely perfect. Feel free just to use the normal map and ignore the "v3" map.

closed 2020-04-23 by WSPeng