StanfordVL/iGibson #35: Environment visualization?

Caselles, 2020-04-21:
Hello, 

I'm having trouble manipulating the PyBullet window representing the simulation.

In this example, I use the Fetch robot in the interactive version of the Placida environment.

[image]

I can zoom on the environment by scrolling, but I did not find any way to move around in this window. Plus, I saw that pressing "G" will show the environment without the additional tabs. 

Can you please indicate what are the relevant commands for this PyBullet window? I would like to move around in the environment freely, in order to visualize it.

Thanks.

ChengshuLi, 2020-04-21:
Thank you for your question. Here are some useful ways to control the pybullet GUI window.

Caselles, 2020-04-22:
Great, thank you so much !

Closing this issue.

closed 2020-04-22 by Caselles