StanfordVL/iGibson #40: ROS Turtlebot navigation launch fails with "model area1" not found error

pirobot, 2020-06-15:
Environment: Ubuntu 16.04, ROS Kinetic, iGibson source install, Python 2.7

I am trying to run the Turtlebot ROS demos as documented here [http://svl.stanford.edu/igibson/docs/ros_integration.html](http://svl.stanford.edu/igibson/docs/ros_integration.html).  The installation went OK but when I try to run one of the demo navigation launch files, I get the error **AssertionError: Model area1 does not exist**.

I'm guessing I need the area1 model.  Is there any way to obtain this model without downloading the entire dataset?  Alternatively, the documentation provides instructions for downloading the Rs model.  Is there a way to run the navigation demo with this model?

Here is the command I am trying and the error it produces:

`roslaunch gibson2-ros turtlebot_gt_navigation.launch`

And the output:

`[ INFO] [1592145720.434257466]: Initializing nodelet with 8 worker threads.
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /home/patrick/src/iGibson/gibson2/assets
INFO:root:Dataset path: /home/patrick/src/iGibson/gibson2/dataset
pybullet build time: Jun  2 2020 06:44:55
******************PyBullet Logging Information:
PyBullet Logging Information******************
Traceback (most recent call last):
  File "/home/patrick/jr2_catkin_ws/src/gibson2-ros/turtlebot_rgbd.py", line 175, in <module>
    node = SimNode()
  File "/home/patrick/jr2_catkin_ws/src/gibson2-ros/turtlebot_rgbd.py", line 47, in __init__
    action_timestep=1 / 30.0)    # assume a 30Hz simulation
  File "/home/patrick/src/iGibson/gibson2/envs/locomotor_env.py", line 53, in __init__
    render_to_tensor=render_to_tensor)
  File "/home/patrick/src/iGibson/gibson2/envs/base_env.py", line 49, in __init__
    self.load()
  File "/home/patrick/src/iGibson/gibson2/envs/locomotor_env.py", line 223, in load
    super(NavigateEnv, self).load()
  File "/home/patrick/src/iGibson/gibson2/envs/base_env.py", line 89, in load
    self.simulator.import_scene(scene, load_texture=self.config.get('load_texture', True))
  File "/home/patrick/src/iGibson/gibson2/core/simulator.py", line 135, in wrapped_load_func
    res = load_func(*args, **kwargs)
  File "/home/patrick/src/iGibson/gibson2/core/simulator.py", line 155, in import_scene
    new_objects = scene.load()
  File "/home/patrick/src/iGibson/gibson2/core/physics/scene.py", line 292, in load
    self.load_floor_metadata()
  File "/home/patrick/src/iGibson/gibson2/core/physics/scene.py", line 125, in load_floor_metadata
    floor_height_path = os.path.join(get_model_path(self.model_id), 'floors.txt')
  File "/home/patrick/src/iGibson/gibson2/utils/assets_utils.py", line 7, in get_model_path
    assert model_id in os.listdir(data_path) or model_id == 'stadium', "Model {} does not exist".format(model_id)
AssertionError: Model area1 does not exist
[turtlebot_gibson_sim-3] process has died [pid 1701, exit code 1, cmd /home/patrick/jr2_catkin_ws/src/gibson2-ros/turtlebot_rgbd.py __name:=turtlebot_gibson_sim __log:=/home/patrick/.ros/log/3398fb82-ae4d-11ea-b2e8-509a4c47197c/turtlebot_gibson_sim-3.log].
log file: /home/patrick/.ros/log/3398fb82-ae4d-11ea-b2e8-509a4c47197c/turtlebot_gibson_sim-3*.log
[ WARN] [1592145725.625318734]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100065 timeout was 0.1.
[ WARN] [1592145730.656886572]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100143 timeout was 0.1.`

frankthecr0c, 2020-07-13:
Hello, I'm trying to use this tool with the same OS and virtual-envs configuration. 
I downloaded the entire gibson_v2 dataset so I can confirm @pirobot  that area1 is available only in this collection.  
Unfortunately I'm experiencing another issue... the area1 subset does not contain the file floor.txt and then I get another **Exceptipon: floors.txt cannot be found in model: area1**

Jingjinganhao, 2020-12-21:
@frankthecr0c 
Hi, I have the same problem as you：
`Exception: floors.txt cannot be found in model: area1`
Do you fix it?

roberto-martinmartin, 2021-06-30:
There was an issue with the assets, the floors.txt file was missing. They should all be there in the latest version.
If you fill out the [form](https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform) and click submit, you should be able to download 2d3ds_for_igibson.zip with the missing floors.txt. Let us know if this works for you and feel free to reopen if you have any other problems!

closed 2021-06-30 by roberto-martinmartin

ChengshuLi, 2021-09-17:
@pirobot @frankthecr0c @Jingjinganhao 
Could you please follow our latest documentation and try again?
http://svl.stanford.edu/igibson/docs/dataset.html#download-gibson-and-stanford-2d-3d-semantics-scenes
You should download 2d3ds_for_igibson.zip. Thanks!