StanfordVL/iGibson #41: Ideal trajectory execution and visualization

rrwiyatn, 2020-06-18:
Hi, is there an example code to generate Figure 7 from iGibson paper (i.e., top down visualization of executed path, shortest path, initial location, and target location)?

Also, is there a way to automatically execute actions that follow the ideal waypoints to collect expert trajectories?

Thanks!

roberto-martinmartin, 2021-06-30:
We are working on this tool, it will be available soon

shurjobanerjee, 2021-08-28:
Hi! I was wondering if there were any updates to this enhancement? Thanks!

genkv, 2022-06-20:
Hi @roberto-martinmartin similar to shurjobanerjee's reply, I was also wondering if there's any update to this work or it's still under development? I just think it would be a great feature to have. thank you