StanfordVL/iGibson #47: Is it possible to get a segmentation mask of interactive objects in the scene?

Caselles, 2020-09-12:
Hello, 

I am looking to modify this script: [examples/demo/env_interactive_example.py](https://github.com/StanfordVL/iGibson/blob/master/examples/demo/env_interactive_example.py) in order to get a segmentation mask of interactive objects (like chairs, sofa, etc) as observation (on top of the RGB observation).

**Is that possible?** What I would like is having the RGB robot view and also a segmentation mask robot view of all the interactive objects that the robot can see in his current field of view.

What I've been trying to do, unsuccessfully:

I've tried adding 'seg' as output in the config file ([examples/configs/turtlebot_interactive_demo.yaml](https://github.com/StanfordVL/iGibson/blob/master/examples/configs/turtlebot_interactive_demo.yaml)) but this is giving me an error: 

`File "[...]/.local/lib/python3.6/site-packages/gibson2/envs/locomotor_env.py", line 323, in get_seg
    if self.num_object_classes is not None:
AttributeError: 'NavigateEnv' object has no attribute 'num_object_classes'`

Caselles, 2020-09-12:
I found more information on your paper and it does seem possible. 

I would like to reproduce exactly what you have in this Figure from your paper: 

[image]

Can you please help me out?

roberto-martinmartin, 2021-06-30:
Duplicated with issue #63

closed 2021-06-30 by roberto-martinmartin