StanfordVL/iGibson #49: Device 0 is not available for rendering

joehays, 2020-09-16:
Hello

I have build the Dockerfile found at: https://github.com/StanfordVL/iGibson/blob/master/docker/headless-gui/Dockerfile

My container is up and running and I have successfully connected the VNC viewer. Using a terminal in the TightVNC Viewer I installed pyTorch with,

`pip install torch' 

then ran the 'env_example.py' and get the following error:

> root@64dc5a86a3e7:/opt/igibson/examples/demo# python env_example.py 
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/assets
INFO:root:Dataset path: /opt/igibson/gibson2/dataset
pybullet build time: Sep  9 2020 17:03:07
INFO:root:Command '['/opt/igibson/gibson2/core/render/mesh_renderer/build/test_device', '0']' returned non-zero exit status 1.
INFO:root:Device 0 is not available for rendering
INFO:root:Device index is larger than number of devices, falling back to use 0
ERROR: Unable to initialize EGL

It's behaving as if it cannot detect the GPU. If I run,

>root@64dc5a86a3e7:/opt/igibson/examples/demo# python -c 'import torch; print(torch.cuda.is_available())'          
True 

pyTorch is able to find the GPU. However, if I run,

>root@64dc5a86a3e7:/opt/igibson/gibson2/core/render/mesh_renderer/build# ./test_device 0
INFO: Unable to initialize EGL     

this utility also fails.  However, 

>root@64dc5a86a3e7:/opt/igibson/gibson2/core/render/mesh_renderer/build# ./query_devices 0
1

returns a '1'

Any thoughts why "env_example.py" is failing to run?

fxia22, 2020-10-14:
Are you starting the docker with `--runtime=nvidia`, I believe that is required to pipe the GPU into the docker container. If it is complaining `Unable to initialize EGL`, you can compare with this [guide](http://svl.stanford.edu/igibson/docs/issues.html).

mjlbach, 2021-07-24:
Hi, we've recently updated the installation guide: http://svl.stanford.edu/igibson/docs/installation.html#system-requirements

The most common reasons we have discovered that reproduce this EGL issue are:

1. There is a bug with cgroups on certain versions of the nvidia 465.x and 450.x drivers that prevents opening an EGL context on all but GPU 0, please update, or have your cluster administrator update your nvidia driver.
2. You are missing libegl-dev (ubuntu/debian package)

mjlbach, 2021-07-27:
Closing because I believe this was addressed with our latest documentation improvements. Please let us know if this is not the case.

closed 2021-07-27 by mjlbach