StanfordVL/iGibson #50: Segmentation fault (core dumped)

ArbiterGe, 2020-09-17:
When I run any program .it  told me "Segmentation fault (core dumped)",Maybe I don't install OpenGL.My service has two GPUs,and If I install openGL i may not login ubuntu

fxia22, 2020-10-14:
Yes, OpenGL installation is required to use iGibson, can you kindly follow the guide here? http://svl.stanford.edu/igibson/docs/issues.html

huaifeng1993, 2020-12-17:
when i ran cmd "ldconfig -p | grep EGL“ ,it show info "
···
libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1
libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so
···
"

zl9501, 2021-01-08:
> when i ran cmd "ldconfig -p | grep EGL“ ,it show info "
> ···
> libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1
> libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
> libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so
> ···
> "

I am in the same situation, and also have "Segmentation fault" error.
How to solve the wrongly linked EGL? Thanks a lot. @fxia22

fxia22, 2021-01-12:
Can you show the results of 1) `nvidia-smi` 2) `glxinfo` (need to install `mesa-utils` first)? Looks like you don't have a OpenGL driver that comes with Nvidia?

Garfield-Finch, 2021-06-16:
I went into this problem, too. Bach then I used WSL2 with a third-party X11 application (VcxSrv). I tackled this using a Ubuntu OS.

ChengshuLi, 2021-09-22:
Please feel free to re-open the issue if you have further questions.

closed 2021-09-22 by ChengshuLi