StanfordVL/iGibson #52: Generating Top-down Semantic Map

shahishikas, 2020-10-05:
Hello, is there a way to get a top-down semantic map of an environment? I'm interested in a map similar to the one below (from the Dataset section of the Docs), but with objects colored semantically.
[image]

yashkant, 2021-02-06:
Hi, I would like to know if this has been possible yet?

ChengshuLi, 2021-09-22:
This script should do the trick. 

```
import cv2
import numpy as np

from igibson.scenes.igibson_indoor_scene import InteractiveIndoorScene
from igibson.simulator import Simulator

def main():
    s = Simulator(mode="headless", image_width=512, image_height=512, device_idx=0)
    scene = InteractiveIndoorScene(
        "Rs_int", texture_randomization=False, object_randomization=False, not_load_object_categories=["ceilings"]
    )
    s.import_ig_scene(scene)
    camera_pose = np.array([0, 0, 8.0])
    view_direction = np.array([0, 0, -1.0])
    s.renderer.set_camera(camera_pose, camera_pose + view_direction, [0, 1, 0])
    seg = s.renderer.render(modes=("seg"))[0]
    seg = (seg[:, :, 0] * MAX_CLASS_COUNT).astype(np.int)
    cv2.imwrite("test.png", seg)

if __name__ == "__main__":
    main()
```

Please feel free to re-open the issue if you have further questions.

mjlbach, 2021-11-12:
I slightly edited eric's example above, and have included it in the next (not yet public) release of iGibson

closed 2021-11-12 by mjlbach