StanfordVL/iGibson #53: Python version

jjerry-k, 2020-10-14:
There is not information about python version. 

When I tested using Python 3.8, this package didn't installed.

Is it worked Only Python 3.6?

fxia22, 2020-10-14:
@jjerry-k  When I tested is also works fine in python3.8. What is the specific issue that you run into?

jjerry-k, 2020-10-14:
> @jjerry-k When I tested is also works fine in python3.8. What is the specific issue that you run into?

Sorry to not post about error message.

It was about cmake. So I tried after installing cmake using apt.

After that, I tried "pip install gibson2" and got this error.

`ERROR: Command errored out with exit status 1:
   command: /home/jerry/miniconda3/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-c04bt5s3/gibson2/setup.py'"'"'; __file__='"'"'/tmp/pip-install-c04bt5s3/gibson2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-iegzbup_
       cwd: /tmp/pip-install-c04bt5s3/gibson2/
  Complete output (997 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.8
  creating build/lib.linux-x86_64-3.8/gibson2
  copying gibson2/__init__.py -> build/lib.linux-x86_64-3.8/gibson2
  creating build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/web_ui.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/semantics_data.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/tf_utils.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/utils.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/assets_utils.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/monitor.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/play.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  copying gibson2/utils/constants.py -> build/lib.linux-x86_64-3.8/gibson2/utils
  creating build/lib.linux-x86_64-3.8/gibson2/learn
  copying gibson2/learn/train_fillers.py -> build/lib.linux-x86_64-3.8/gibson2/learn
  copying gibson2/learn/vision_utils.py -> build/lib.linux-x86_64-3.8/gibson2/learn
  copying gibson2/learn/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/learn
  copying gibson2/learn/completion.py -> build/lib.linux-x86_64-3.8/gibson2/learn
  creating build/lib.linux-x86_64-3.8/gibson2/external
  copying gibson2/external/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/external
  creating build/lib.linux-x86_64-3.8/gibson2/core
  copying gibson2/core/simulator.py -> build/lib.linux-x86_64-3.8/gibson2/core
  copying gibson2/core/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/core
  creating build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/demo_interactive.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/base_env.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/demo.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/challenge.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/parallel_env.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  copying gibson2/envs/locomotor_env.py -> build/lib.linux-x86_64-3.8/gibson2/envs
  creating build/lib.linux-x86_64-3.8/gibson2/data_utils
  copying gibson2/data_utils/visualize_path.py -> build/lib.linux-x86_64-3.8/gibson2/data_utils
  copying gibson2/data_utils/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/data_utils
  copying gibson2/data_utils/datasets.py -> build/lib.linux-x86_64-3.8/gibson2/data_utils
  copying gibson2/data_utils/utils.py -> build/lib.linux-x86_64-3.8/gibson2/data_utils
  copying gibson2/data_utils/generate_data.py -> build/lib.linux-x86_64-3.8/gibson2/data_utils
  creating build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/kuka_primitives.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/pr2_primitives.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/pr2_never_collisions.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/transformations.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/pr2_utils.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/voxels.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/parse_json.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/utils.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  copying gibson2/external/pybullet_tools/pr2_problems.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools
  creating build/lib.linux-x86_64-3.8/gibson2/external/motion
  copying gibson2/external/motion/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/external/motion
  creating build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools/ikfast
  copying gibson2/external/pybullet_tools/ikfast/__init__.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools/ikfast
  copying gibson2/external/pybullet_tools/ikfast/utils.py -> build/lib.linux-x86_64-3.8/gibson2/external/pybullet_tools/ikfast
[... cut ...]
    File "/tmp/pip-install-c04bt5s3/gibson2/setup.py", line 97, in build_extension
      subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
    File "/home/jerry/miniconda3/lib/python3.8/subprocess.py", line 364, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j2']' returned non-zero exit status 2.
  ----------------------------------------
  ERROR: Failed building wheel for gibson2`

fxia22, 2020-10-14:
Thanks for the detailed log. Looks like the issue is that you don't have opengl headers installed. You would need to install `GL/gl.h`, it is in:
```
sudo apt install mesa-common-dev
```

closed 2020-10-14 by jjerry-k