StanfordVL/iGibson #56: Objects are not rendered properly in robot view and external view

XavierBouvard, 2020-12-11:
Hello,

Since last update on code base the rendering of object seems buggy.
when running:

```bash
cd ./examples/demo/
python simulator_example.py
```

the cracker boxes are black in the rendered images
[image]

Did I missed something or can you reproduce ?  
Can you point me in the right direction to fix it ?  

FYI: I implemented my own type of objects to include images in the environment (ex: apriltags) with the same result. Since I haven't pull iGibson for a while, I cannot tell you which commit is to incriminate but I'd be happy to help you on that if it can help.

best regards,

fxia22, 2020-12-11:
I cannot seem to reproduce with the HEAD of the master branch and my rendering looks like this:
[image]

I would suggest you check a few things:

- is the build latest? You can run `./clean.sh` and `pip install -e .` to build again.
- are the YCB assets up to date? You can remove the assets folder and run `python -m gibson2.utils.assets_utils --download_assets` again. 
- does the ycb objects show correct textures in the pybullet view?

fxia22, 2020-12-11:
I figured out the reason, the environment light probes (which are used to light the scene) come with iG Dataset are not properly loaded so that the cracker box is not lit, you can solve it with either of the below methods:

1.  Download the full iG dataset, by running `python -m gibson2.utils.assets_utils --download_ig_dataset` [ref](http://svl.stanford.edu/igibson/docs/dataset.html)
2. Don't use environment lighting for rendering objects, by changing the rendering setting in `simulator_example.py` to `settings = MeshRendererSettings(enable_shadow=False, msaa=False, enable_pbr=False)`

XavierBouvard, 2020-12-11:
Thanks a lot for your quick response.

changing the enable_pbr to false did fix the issue:
[image]

thanks again

closed 2020-12-11 by XavierBouvard