StanfordVL/iGibson #58: Install iGibson from source  in python2.7

Jingjinganhao, 2020-12-17:
Hi,
I install iGibson from source following installation guide in python2.7. However, when I run
`roslaunch gibson2-ros turtlebot_rgbd.launch`
It shows:
` File "/home/cethik/yejingjing/iGibson/gibson2/simulator.py", line 314
    if (filename, (*dimensions)) not in self.visual_objects.keys():
                                ^
    SyntaxError: invalid syntax
`
Do you have the source code of igibson for python 2

fxia22, 2020-12-22:
Thanks for reporting, we are working on a fix.

Jingjinganhao, 2020-12-23:
@fxia22 
Hi, I run `pip install gibson2 ` in my python2 environment， it shows
`with open("README.md", "r", encoding="utf-8") as fh:
    TypeError: 'encoding' is an invalid keyword argument for this function`
It seems the setup.py  is suitable for python3.
Then I download iGibson-0.0.4.zip， when I run `pip install -e .`
It shows
`iGibson-0.0.4/gibson2/core/render/pybind11 
      does not contain a CMakeLists.txt file.
`
Finally, I download gibson2-0.0.4-cp27-cp27mu-manylinux1_x86_64.whl, and pip install it.
copy site-packages/gibson2/core/render/pybind11 to /iGibson-0.0.4/gibson2/core/render/pybind11 
then run `pip install -e .` ,it works

fxia22, 2020-12-24:
Python2.7 building issue is solved in the latest version v1.0.1 and you should be able to install it with python 2.7.

closed 2021-01-05 by fxia22