StanfordVL/iGibson #59: Exception: floors.txt cannot be found in model: area1

Jingjinganhao, 2020-12-22:
Hi，
there is something wrong with me，when  I run 
`roslaunch gibson2-ros turtlebot_rgbd.launch`  
it shows:
  `Exception:floors.txt cannot be found in model: area1` 
   I have downloaded the entire gibson_v2 dataset, and the area1 subset does not contain the file floors.txt.
   How can I get floors.txt?

fxia22, 2020-12-24:
Hi @Jingjinganhao, you might have downloaded the wrong dataset, can you download again from https://forms.gle/36TW9uVpjrE1Mkf9A and look for 2d3ds_for_igibson.zip? Thanks.

Jingjinganhao, 2020-12-25:
@fxia22 
When I run
`python -m gibson2.utils.assets_utils --download_dataset https://***/2d3ds_for_igibson.zip
`
it shows 
`正在连接 storage.googleapis.com (storage.googleapis.com)|172.217.24.16|:443... 已连接。
已发出 HTTP 请求，正在等待回应... 403 Forbidden
2020-12-25 11:11:49 错误 403：Forbidden。

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
`
And I can't open the link of 2d3ds_for_igibson.zip with a browser.
But I can download other datasets.
Is there something wrong with the link of  2d3ds_for_igibson.zip?

fxia22, 2020-12-26:
@Jingjinganhao  sorry, there was something wrong with permission. You can try again now.

Jingjinganhao, 2020-12-28:
@fxia22 
Thanks and sorry to bother you again.
I download the dataset and run `roslaunch gion2-ros turtlebot_rgbd.launch `, it shows:
`lidar_points = obs['scan']`
`KeyError: 'scan'`
also in rviz,  it shows:
`Fixed Frame [odom] does not exist`
and can't load turtlebot robot.
Is there something wrong with the turtlebot.urdf? Missing odom?

fxia22, 2021-01-05:
Hi @Jingjinganhao, this issue is recently fixed in the master branch, can you check if it works for you now? We also updated the instruction page here: http://svl.stanford.edu/igibson/docs/ros_integration.html

Jingjinganhao, 2021-01-08:
Hi @fxia22 , that works for me . But I have other problems.
When I run `roslaunch gibson2-ros turtlebot_gmapping.launch `
I can't make the robot move through the keyboard.
When I change `<remap from="turtlebot_teleop_keyboard/cmd_vel" to="/cmd_vel"/> ` in keyboard_teleop.launch to `<remap from="turtlebot_teleop_keyboard/cmd_vel" to="/mobile_base/commands/velocity"/> `, it works.
and I can't get map in rviz also.
When I change `<remap from="scan" to="/gibson_ros/laserscan"/>` in turtlebot_gmapping.launch to `<remap from="scan" to="/scan"/>`, it works.
Maybe there is something wrong with remap.

fxia22, 2021-01-08:
Hi @Jingjinganhao, 

I am glad the rest of the pipeline works. We will fix the remapping problems. Feel free to reach out if you spot other issues with ros.

Jingjinganhao, 2021-01-08:
Hi @fxia22 ,
sorry to bother you again, 
In [http://svl.stanford.edu/igibson/docs/ros_integration.html](url) ,  when running the gmapping example, you can see the third view of the camera by bullet. How I can get it?
and when I run 
` roslaunch gibson2-ros turtlebot_gt_navigation.launch `
or
`roslaunch gibson2-ros turtlebot_navigation.launch `
or
`roslaunch gibson2-ros turtlebot_hector_mapping.launch `
It shows
`No such file or directory: u'/home/**/catkin_ws/src/turtlebot/turtlebot_teleop/launch/ps3_teleop.launch`
There are  only keyboard_teleop_cpp.launch and keyboard_teleop.launch in/turtlebot_teleop/launch/

Jingjinganhao, 2021-01-08:
Hi @fxia22 ,
is there any examples about training a RL model by using ros?

fxia22, 2021-01-12:
Hi @Jingjinganhao, if you would like to use ps3 controller to teleoperate the robot, you can grab the ps3_teleop.launch from here: https://github.com/turtlebot/turtlebot/tree/melodic/turtlebot_teleop/launch

Feel free to change it to keyboard if you don't have a ps3 controller. 

In order to get the bullet view, you can change here https://github.com/StanfordVL/iGibson/blob/master/gibson2/examples/ros/gibson2-ros/turtlebot_rgbd.py#L50
from `headless` to `gui`.

> re: is there any examples about training a RL model by using ros?

In iGibson's case, ROS is supported for evaluating traditional planning based methods that can be found in ros' repository. If you want to use RL, you don't need to use it with ROS. We have a tutorial of using iGibson with tf-agents here: http://svl.stanford.edu/igibson/docs/learning_framework.html

Jingjinganhao, 2021-01-15:
Okay, thank you!

Jingjinganhao, 2021-01-15:
Okay, thank you !@fxia22

YantianZha, 2021-01-24:
> @fxia22
> When I run
> `python -m gibson2.utils.assets_utils --download_dataset https://***/2d3ds_for_igibson.zip `
> it shows
> `正在连接 storage.googleapis.com (storage.googleapis.com)|172.217.24.16|:443... [...]

Hi @fxia22:
I have the same problem as @Jingjinganhao of downloading the 2d3ds_for_igibson.zip from the [link](https://forms.gle/36TW9uVpjrE1Mkf9A) you gave. I cannot find any downloadable link to get the 2d3ds_for_igibson.zip. I don't know if it means we only need to download the ig_dataset.tar.gz which is just above the 2d3ds_for_igibson.zip? But I cannot find the floors.txt file in ig_dataset.tar.gz (so I suppose I still need to download 2d3ds_for_igibson.zip). Thank you!

ChengshuLi, 2021-04-13:
If you fill out the [form](https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform) and click submit, you should be able to download 2d3ds_for_igibson.zip.

roberto-martinmartin, 2021-06-30:
The latest version should include the missing files. Feel free to reopen if the issue was not solved

closed 2021-06-30 by roberto-martinmartin

AaronWaxson, 2021-07-05:
I downloaded the 2d3ds_for_igibson.zip latest version, and solve the 'floors.txt' exception.
But I get a new one that is:
`Exception: floor_heights.txt cannot be found in mode: area1`
When I run `roslaunch gibsion-ros turtlebot_rgbd.launch`
Any idea about that ? I run in python2.7.

AaronWaxson, 2021-07-05:
> The latest version should include the missing files. Feel free to reopen if the issue was not solved

I downloaded the 2d3ds_for_igibson.zip latest version, and solve the 'floors.txt' exception.
But I get a new one that is:
Exception: floor_heights.txt cannot be found in mode: area1
When I run roslaunch gibsion-ros turtlebot_rgbd.launch
Any idea about that ? I run in python2.7.

AaronWaxson, 2021-07-05:
> > The latest version should include the missing files. Feel free to reopen if the issue was not solved
> 
> I downloaded the 2d3ds_for_igibson.zip latest version, and solve the 'floors.txt' exception.
> But I get a new one that is:
> Exception: floor_heights.txt cannot be found in mode: area1
> When I run roslaunch gibsion-ros turtlebot_rgbd.launch
> Any idea about that ? I run in Ubuntu 16, python2.7.

And I downloaded gibson_fullplus dataset,  and after I unzipped it. I didn't see the dataset format like document told. I only have that

Is there any problem? Help plz.

elhamAm, 2021-08-17:
I am having the exact same problem for the Matterport dataset

ChengshuLi, 2021-09-17:
@AaronWaxson @elhamAm replied to you in the other issue.