StanfordVL/iGibson #60: ”list index out of range“  when process a CubiCasa5K scene

zhxsking, 2021-01-14:
Hello,

I want to process a CubiCasa5K scene, but i got this:

`root@43a153b9efd4:/opt/igibson/gibson2/utils/data_utils/ext_scene# ./scripts/process_scene_cubicasa.sh /home/zhaoxin/data/cubicasa5k/colorful/3006/
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/zhaoxin/data/ig_dataset-1.0/
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/zhaoxin/data/ig_dataset-1.0/
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
/home/zhaoxin/data/cubicasa5k/colorful/3006
3006
Blender 2.82 (sub 7) (hash 77d23b0bd76f built 2020-02-12 17:14:50)
Read prefs: /root/.config/blender/2.82/config/userpref.blend
found bundled python: /opt/igibson/gibson2/utils/data_utils/blender_utils/blender-2.82-linux64/2.82/python
/opt/igibson/gibson2/data/cubicasa_dataset/scenes/3006_floor_0
log out
Error: File format is not supported in file '/opt/igibson/gibson2/data/cubicasa_dataset/scenes/3006_floor_0'

Blender quit
----------------3------------------
----------------4------------------
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/zhaoxin/data/ig_dataset-1.0/
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
pybullet build time: Dec 23 2020 01:46:51
all categories: ['washer', 'toilet', 'bottom_cabinet', 'sink', 'top_cabinet', 'bench', 'fridge', 'stove', 'dishwasher', 'window', 'door']
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/zhaoxin/data/ig_dataset-1.0/
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
pybullet build time: Dec 23 2020 01:46:51
all categories: ['washer', 'toilet', 'bottom_cabinet', 'sink', 'top_cabinet', 'bench', 'fridge', 'stove', 'dishwasher', 'window', 'door']
----------------5------------------
3006_floor_0
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/zhaoxin/data/ig_dataset-1.0/
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
pybullet build time: Dec 23 2020 01:46:51
torch is not available, falling back to rendering to memory(instead of tensor)
CUBICASA
INFO:root:IndoorScene model: 3006_floor_0
INFO:root:StaticIndoorScene scene: 3006_floor_0
Traceback (most recent call last):
  File "step_5_generate_trav_map.py", line 80, in <module>
    main()
  File "step_5_generate_trav_map.py", line 76, in main
    generate_trav_map(scene_name, args.source, False)
  File "step_5_generate_trav_map.py", line 38, in generate_trav_map
    scene_source=scene_source)
  File "/opt/igibson/gibson2/scenes/igibson_indoor_scene.py", line 120, in __init__
    self.load_room_sem_ins_seg_map(seg_map_resolution)
  File "/opt/igibson/gibson2/scenes/igibson_indoor_scene.py", line 340, in load_room_sem_ins_seg_map
    sem_name = room_cats[sem_id - 1]
IndexError: list index out of range
`

Thank you so much!

ChengshuLi, 2021-01-14:
@fxia22 This is probably fixed by this commit. Could you update the asset? https://github.com/StanfordVL/ig_dataset/commit/3c559888c7c1a2585d96edd64f8d6d9755ee1926

fxia22, 2021-01-14:
@ChengshuLi will do.

fxia22, 2021-01-15:
@zhxsking I have updated the assets, can you download and try again? The new version have version tag 1.0.1

`wget https://storage.googleapis.com/gibson_scenes/ig_dataset.tar.gz`

zhxsking, 2021-01-18:
@fxia22 thx，i try it again and now it works!

fxia22, 2021-01-18:
great, thanks.

closed 2021-01-18 by fxia22