StanfordVL/iGibson #61: Failing motion planning test

DanielaPamplona, 2021-02-02:
Hi, 
I installed iGibson with pip on Ubuntu 20 and when running the tests the motion planning module fails:
================================= short test summary info =============================
FAILED test_motion_planning.py::test_base_planning - TypeError: object of type 'NoneType' has no len()
===================1 failed, 1 passed, 7 warnings in 25.33s========================================

I manage to track the error, using the debugger pdb and noticed that collision_fn(end_conf)  is true in plan_base_motion_2d of /gibson2/external/pybullet_tools/utils.py 
Which makes that the plan is Null and the test fails in test_motion_planning.py  
To test if it was a problem with the initial position, I changed the initial position values, and the error persists, I also changed the plan base motion, and the error is still there. 

Any hints?

fxia22, 2021-02-02:
This suggests a plan is not found. Sampling based motion planner does have a probability of returning no plans. Does this issue persist if you run multiple times?

DanielaPamplona, 2021-02-02:
Tks for the prompt response!
I just run 10 times and every time it failed in the same place.

fxia22, 2021-02-03:
The occupancy grid should be 0,0.5 and 1 meaning free space(1), unknown space(0.5), and obstacles(0). The occupancy grid in the test case should look like the following, are you able to get the same occupancy grid?

[image]

DanielaPamplona, 2021-02-03:
The occupancy grid is similar to what you have shared.
But, I noticed that when I run the single test:
pytest test_motion_planning.py::test_base_planning
the test runs without errors, while with:
pytest test_motion_planning.py
The test fails in the same function.

Actually, this is a general error with pytest. If I type:
pytest --ignore disabled --ignore benchmark
it fails most of the tests, and it takes ages to run. But if I type:
for f in *.py; do  
pytest --ignore disabled --ignore benchmark $f; 
done 
It passes the tests, except for the motion planning.
Anyway, the problem is solved, although I could not understand why it was falling.
Thanks for your help.

ChengshuLi, 2021-04-13:
Feel free to re-open the issue if you have further questions!

closed 2021-04-13 by ChengshuLi

sycz00, 2022-02-24:
Hey @ChengshuLi 
Iam having a follow up question. Is it possible to simulate the trajectory taken by the motion planner ? More specifically, Iam creating a map during each step and while using the motion planner, the agent gets just teleported to the end point.

ChengshuLi, 2022-02-25:
Thanks for the question. Currently, we don't have a good way to physically simulate the trajectory taken by the motion planner. You can potentially teleport the robot to each waypoint on the path, but it's still not physically simulating the robot. The best thing to do would be to actuate the robot's wheels to follow the waypoints with closed-loop control, but it will take some time to implement this feature. Contribution is very welcome!

cgokmen, 2022-02-25:
As a side note, this feature is in the works for our next release, but I'm hesitant to give a concrete timeline.