StanfordVL/iGibson #62: GLSL 1.5.0 is not supported

sanjeevkoppal, 2021-02-10:
Hi,

I followed the instructions for Gibson2 installation. When I run the demo code test:

I get this error:
GLSL 1.5.0 is not supported. Supported versions are ....

I did retry the installation with USE_GLAD set to FALSE in CMakeLists, but this resulted in the installation crashing.

Any ideas on the next steps I can take?

fxia22, 2021-02-10:
Can you provide a bit more information about your platform? (os, nvidia driver version, etc)

Can you also show the output of `glxinfo`? Thanks!

fxia22, 2021-02-10:
Judging from a few posts I read online a potential solution would be to add environment variable
`MESA_GL_VERSION_OVERRIDE=3.2` or `MESA_GL_VERSION_OVERRIDE=3.3`, not sure what is the intuition behind it.

source:
https://github.com/OpenHantek/openhantek/pull/319
https://discourse.slicer.org/t/error-glsl-1-50-is-not-supported/10083

sanjeevkoppal, 2021-02-10:
Hi. 

I  am on Linux, specifically  a Red Hat Enterprise Server. 
I am using an NVIDIA GeForce GPU with 440.64  version driver  and cuda version 10.2 (those numbers from nvidia-smi)
When I  do ldcongif -p | grep EGL I can see 
[image]

When I run glxinfo, it gives an unknown command error.  Would you happen to know where the command is located?

fxia22, 2021-02-10:
glxinfo is in `mesa-utils` in ubuntu, https://packages.ubuntu.com/search?keywords=mesa-utils

I am not sure where it is on redhat. 

judging from your `ldconfig` it looks fine because you have libEGL_nvidia installed.

sanjeevkoppal, 2021-02-10:
Do you know where I could make that change?

[External Email]

OpenHantek/openhantek#319<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_OpenHantek_openhantek_pull_319&d=DwMCaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=UTCUHUgNwObcuZsLsLMIiez54sUEmW8TukUlmQVuvEY&m=QVi4ImGVlKUvOUj3VyeaC68gMQvkk_MYXuKVyI-tQAA&s=V-7bop6mSfoXB2j19bKuGskWbBqbLfJlHa5egjSKH04&e=>
https://discourse.slicer.org/t/error-glsl-1-50-is-not-supported/10083<https://urldefense.proofpoint.com/v2/url?u=https-3A__discourse.slicer.org_t_error-2Dglsl-2D1-2D50-2Dis-2Dnot-2Dsupported_10083&d=DwMCaQ&c=sJ6xIWYx-zLMB3EPkvcnVg&r=UTCUHUgNwObcuZsLsLMIiez54sUEmW8TukUlmQVuvEY&m=QVi4ImGVlKUvOUj3VyeaC68gMQvkk_MYXuKVyI-tQAA&s=kXF3KZ8m9OhtSMRXZUMSGCqY5sdlTEattFkuh6vsXNw&e=>

sanjeevkoppal, 2021-02-10:
> Judging from a few posts I read online a potential solution would be to add environment variable
> `MESA_GL_VERSION_OVERRIDE=3.2` or `MESA_GL_VERSION_OVERRIDE=3.3`, not sure what is the intuition [...]

Do you know where I would edit this?

sanjeevkoppal, 2021-02-10:
I just want to say that I ran the following on the Linux command line:
export MESA_GL_VERSION_OVERRIDE=3.3

Now the error does not show up. For a split second I can see the environment, but then it seg faults.

ChengshuLi, 2021-02-10:
Which example code are you trying to run?

fxia22, 2021-02-10:
I guess it should not be trying to use openGL from mesa in the first place. Can you check the output of `glxinfo`, that would be the most informative. Other debug guides on this issue can be found in http://svl.stanford.edu/igibson/docs/issues.html

sanjeevkoppal, 2021-02-10:
I tried both env_example and demo_static. Here is the output from demo_static:
[image]

fxia22, 2021-02-10:
Ok, this is very helpful, it is showing you are using mesa as openGL renderer, but it should be using nvidia openGL, otherwise you are not using the GPU. It would be good to figure out why nvidia opengl is not used.

Are you operating locally or trying to connect over ssh?

sanjeevkoppal, 2021-02-10:
I am operating over an interactive desktop run by TurboVNC (setup by the server management). I start this session on a GPU.

I understood what you said about using nvidia openGL. I am not sure about how to set that up, but I will investigate.

sanjeevkoppal, 2021-02-10:
Been investigating, but no luck. You can see from my previous posts that both Mesa and Nvidia openGL are installed, and yet I cannot get it to point only to the Nvidia open gl. Any suggestions?

sanjeevkoppal, 2021-02-10:
Also, glxinfo is showing no command found.

sanjeevkoppal, 2021-02-10:
Folks, believe it or not, it started working. I changed multiple things at once, so I am not sure what it was:
1. Added an ubuntu linux container module
2. Increased GPU memory (my bet is on this)
3. So far the static has worked.
Thanks for your assistance.
[image]

closed 2021-02-10 by sanjeevkoppal