StanfordVL/iGibson #63: segmentation returning three single values

DanielaPamplona, 2021-02-12:
Hi,
I was trying to render with the segmentation mode in the robot camera renderer. 
First I changed in the yaml file to
output: [rbg, seg]
and then I wrote on the main file
imgs = sim.renderer.render_robot_cameras(modes=('rgb', 'seg'))

imgs is a list of two arrays. The first array is what I expected. But the second array is strange. 
It as shape as the image, (256, 256, 4), but it has only 3 values:
imgs[1][:,;0]=0.01176471
imgs[0][:,:,1] = 0
imgs[0][:,:,2] = 0
imgs[0][:,:,3] = 1

What am I missing? Can you make a simple demo with the segmentation? :D (I know it is a bit too much to ask this, but maybe it is not that much work for you...)

DanielaPamplona, 2021-02-15:
Following this, I just changed the gibson2/examples/demo/mesh_renderer_simple_example.py to have three rendering modes:
modes=('rgb', 'normal', 'seg') and the output is this black.

[image]

The output on the shell does not say refer to any error or at least I cannot see it...
--------------------------------------------------------------------------------------------------------------------------

INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/assets
INFO:root:Gibson Dataset path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/cubicasa_dataset
INFO:root:Example path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/examples
INFO:root:Example config path: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/examples/configs
pybullet build time: Feb  2 2021 00:18:07
INFO:root:Device 0 is available for rendering
INFO:root:Command '['/home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:root:Device 1 is not available for rendering
INFO:root:Using device 0 for rendering
Compiling GLSL shader: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/shaders/450/equirect2cube_cs.glsl
Compiling GLSL shader: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/shaders/450/spmap_cs.glsl
Compiling GLSL shader: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/shaders/450/irmap_cs.glsl
Compiling GLSL shader: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/shaders/450/spbrdf_cs.glsl
Loading image: /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/ig_dataset/scenes/background/photo_studio_01_2k.hdr
INFO: compiled pbr shaders
INFO:root:Loading /home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/data/g_dataset/Rs/mesh_z_up.obj
/home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:461: RuntimeWarning: divide by zero encountered in true_divide
  r = 1.0 / (delta_uv1[:, 0] * delta_uv2[:, 1] -
/home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:463: RuntimeWarning: invalid value encountered in multiply
  tangent = (delta_pos1 * delta_uv2[:, 1][:, None] -
/home/louisette/anaconda3/envs/py3-igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:465: RuntimeWarning: invalid value encountered in multiply
  bitangent = (delta_pos2 * delta_uv1[:, 0][:, None] -

-----------------------------------------------------------------------
Any idea of the problem?

shrirangsp, 2021-04-07:
Hi,

Were you able to find solution for this issue #47 ?

DanielaPamplona, 2021-04-09:
No, I was not, I just gave up and used  [Robotrix](https://github.com/3dperceptionlab/therobotrix)

fxia22, 2021-04-09:
Hi @shrirangsp and @DanielaPamplona , sorry for the delay. The example shown above is rendering a Gibson scene (not iGibson), so it doesn't come with semantic segmentation. You can only differentiate added objects and background mesh.

However, if you load an iGibson scene, it comes with full semantic segmentations, here is an example:

```python
from gibson2.envs.igibson_env import iGibsonEnv
from gibson2.utils.motion_planning_wrapper import MotionPlanningWrapper
from gibson2.objects.ycb_object import YCBObject
import argparse
import numpy as np
import gym
from gibson2.scenes.igibson_indoor_scene import InteractiveIndoorScene
import matplotlib.pyplot as plt
from tqdm import tqdm
mode = 'headless'
config = '/home/fei/Development/gibsonv2/gibson2/examples/configs/fetch_room_rearrangement.yaml'
nav_env = iGibsonEnv(config_file=config,
                     mode=mode,
                     action_timestep=1.0 / 120.0,
                     physics_timestep=1.0 / 120.0)

nav_env.reset()
seg = nav_env.simulator.renderer.render_robot_cameras(modes=('seg'))[0]
print(np.unique(seg[:,:,0]))

plt.figure()
plt.imshow(seg[:,:,0])
plt.show()

````

You will see something like this:

[image]

The labels can be parsed with 

```python
from gibson2.utils.semantics_utils import get_class_name_to_class_id
print(get_class_name_to_class_id())
```

```
{'armchair': <SemanticClass.SCENE_OBJS: 3>,
 'basket': 4,
 'bathtub': 5,
 'bed': 6,
 'bench': 7,
 'bottom_cabinet': 8,
 'bottom_cabinet_no_top': 9,
 'breakfast_table': 10,
 'burner': 11,
 'carpet': 12,
 'chaise_longue': 13,
 'chest': 14,
 'coffee_maker': 15,
 'coffee_table': 16,
 'console_table': 17,
 'countertop': 18,
 'crib': 19,
 'desk': 20,
 'dishwasher': 21,
 'door': 22,
 'dryer': 23,
 'floor_lamp': 24,
 'folding_chair': 25,
 'fridge': 26,
 'gaming_table': 27,
 'grandfather_clock': 28,
 'guitar': 29,
 'heater': 30,
 'highchair': 31,
 'laptop': 32,
 'loudspeaker': 33,
 'microwave': 34,
 'mirror': 35,
 'monitor': 36,
 'oven': 37,
 'pedestal_table': 38,
 'piano': 39,
 'picture': 40,
 'pillow': 41,
 'pool_table': 42,
 'pot_plant': 43,
 'rail_fence': 44,
 'range_hood': 45,
 'rocking_chair': 46,
 'shelf': 47,
 'shower': 48,
 'sink': 49,
 'sofa': 50,
 'speaker_system': 51,
 'stand': 52,
 'standing_tv': 53,
 'stool': 54,
 'stove': 55,
 'straight_chair': 56,
 'swivel_chair': 57,
 'table_lamp': 58,
 'toilet': 59,
 'top_cabinet': 60,
 'towel_rack': 61,
 'trash_can': 62,
 'treadmill': 63,
 'wall_clock': 64,
 'wall_mounted_tv': 65,
 'washer': 66,
 'window': 67}
```

Hope this is helpful!

sycz00, 2021-04-30:
Hey @fxia22  

is it somehow possible to add a segmentation mask for user added objects such that individual objects are segmented as well and not just a single class for "user_added_objects" ?

ChengshuLi, 2021-04-30:
Yes, you would just need to specify different class ids when calling `simulator.import_object`. By default, it's just a single class for "user_added_objects". 

https://github.com/StanfordVL/iGibson/blob/e5029110de6d619b4d66bf9a91bb90d5ea24ff8d/gibson2/simulator.py#L247-L252

sycz00, 2021-04-30:
@ChengshuLi 
Thank you, that must have slipped through my eyes...

closed 2021-06-30 by roberto-martinmartin