StanfordVL/iGibson #65: why is building scene more realistic in previous version "GIbsonEnv"

nanlliu, 2021-03-08:
im just wondering why "GibsonEnv"'s "building_scene.py" can make more realistic scene in "RS_interactive"? And how can i do the same thing in iGIbson?

i tried "GibsonEnv" but it has some issues when running 'build local".

much appreciate it

fxia22, 2021-03-09:
Hi Nan,

We are fully backward compatible with Gibson Env scenes, one example can be found here:
https://github.com/StanfordVL/iGibson/blob/master/gibson2/examples/demo/scene_example.py

The scene looks like the following: 
[image]

Is this what you are looking for? In general, we support scanned meshes as scenes (non-interactive), and synthetic scenes (that generated from annotating scanned meshes) that are fully interactive.

nanlliu, 2021-03-09:
much appreciate for your reply!

i probably didn't convey what i was trying to ask. sorry about that!

i think in version 0.0.4, "gibson2.core.physics.scene" has "BuildingScene" object. When i used it with "Rs_interactive", it was a bit more realistic than this one. im just wondering if that is removed in the latest version?

fxia22, 2021-03-09:
Hi Nan,

Now I understand your question. From v0.0.4 to v1.0+ we have switched the rendering pipeline. Previously the rendering is using ray-tracing based texture baking and rendering albedo only, this causes the shadows to present even if you move the objects away. Now we are using physically based rendering with shadow mapping, so the shadows are dynamically generated. The lighting effect might not be as rich as v0.0.4, but the benefit is the rendering of dynamic objects is much better.

nanlliu, 2021-03-09:
i see. appreciatae that!

nanlliu, 2021-03-09:
and just a heads up!

when using simulator to render images, when "render_to_tensor" is enabled, the image will be saved upside down compared with the flag being not set.

ChengshuLi, 2021-04-13:
Feel free to re-open the issue if you have further questions!

closed 2021-04-13 by ChengshuLi