StanfordVL/iGibson #66: Motion Planning Wrapper - Arm Motion Planning

shrirangsp, 2021-03-09:
Hello,

1) Can you please provide a example for arm motion planning using the MotionPlanningWrapper?
2) Is MotionPlanningWrapper only compatible with iGibsonEnv? Because I want to use it in a StadiumScene.

fxia22, 2021-03-10:
Hi @shrirangsp ,

You can run the motion planning example in `gibson2/examples` with `python motion_planning_example.py -m iggui -c ../configs/fetch_room_rearrangement.yaml`.

The instruction for using the viewer to do the motion planning can be found here: http://svl.stanford.edu/igibson/docs/viewer.html

And the relevant code is here (plan and execute a push motion): 

https://github.com/StanfordVL/iGibson/blob/master/gibson2/render/viewer.py#L426-L435

shrirangsp, 2021-03-18:
Thank you for your reply. That helped me understand how to use the motion planning wrapper. 
**_Query_** -  I want to create a simple grasping task to grab a object using robots end-effector, in a StadiumScene. But it seems that MotionPlanningWrapper is not compatible with the stadium scene, is there any alternative approach that you can suggest for doing a grasping task.

shrirangsp, 2021-03-19:
Thank you for your reply. 
**_Query_** - I want to create a pick and place task using MotionPlanninngWrapper. For this, do I need to create a new custom task?

closed 2021-03-25 by shrirangsp