StanfordVL/iGibson #67: Can we get a voxel grid?

teamiceberg, 2021-03-11:
After setting up a scene - let’s say random objects on a shelf, we would like to get a voxel grid of the scene with the entire scenes made up of voxels of certain size we would want. Is there a built-in function that allows us to generate such a voxel grid?

thank you.

ChengshuLi, 2021-03-11:
Hi, we don't have a built-in function for that. However, one thing you can try is to create a `p.GEOM_BOX` in pybullet of certain size (your voxel grid resolution, say 5cm x 5cm x 5xm), and then move this box around in the scene. At each position, call `p.stepSimulation` and then call `p.getContactPoints`to detect collision. If there is a collision, then that voxel cell is occupied, and vice versa.

fxia22, 2021-03-11:
We have a utility function to dump all the vertices and faces from the renderer (a snapshot of the scene at a given moment), https://github.com/StanfordVL/iGibson/blob/master/gibson2/render/mesh_renderer/mesh_renderer_cpu.py#L875-L891

You can pipe the output to some voxelization method to generate voxels of the scene, e.g. http://www.open3d.org/docs/latest/tutorial/Advanced/voxelization.html

ChengshuLi, 2021-03-11:
Forgot to add. 
Before the for loop of moving the box around, you need to do `p.saveState()`
After each `p.getContactPoints`, you need to do `p.restoreState()`

Something like this
```
        state_id = p.saveState()
        for each position in 3D space:
            p.resetBasePositionAndOrientation(
                body_id, position, [0, 0, 0, 1])
            p.stepSimulation()
            if len(p.getContactPoints(body_id)) > 0:
                # occupied
                voxel_grid[map_pos[0], map_pos[1]] = 1
            p.restoreState(state_id)
```

Probably Fei's method is more efficient though. Feel free to try that first.

shivani1494, 2021-04-05:
out of curiosity, did you have to dig into bullet code to figure out whether the collision points returned are vertices of voxelized object or vertices on the object mesh contained in .obj file? or do you do some preprocessing in iGibson? I am a similar use case.

shivani1494, 2021-04-05:
have a similar use case- did you have to dig into bullet code to figure out whether the collision points returned are vertices of voxelized object or vertices on the object mesh passed through .obj file? or do you do some preprocessing in iGibson? @ChengshuLi  
let me know if I am unclear.

ChengshuLi, 2021-04-13:
Feel free to re-open the issue if you have further questions!

closed 2021-04-13 by ChengshuLi