StanfordVL/iGibson #68: Invalid Render Device Issue

betrevisan, 2021-03-15:
Hi!

After installing iGibson using pip, we try to run the demo using `python -m gibson2.examples.demo.demo_static`. However, we keep getting an error that says `ERROR: Invalid render_device choice: 0 < 0.`. We have two CPU cores requested with 1 GPU per core. Additionally, we have also checked and EGL is not linked to mesa. Has anyone else encountered the same problem? Here is what we have on the terminal:

`Call stack:
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/examples/demo/demo_static.py", line 33, in <module>
    DemoStatic().run_demo()
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/examples/demo/demo_static.py", line 19, in run_demo
    s = Simulator(mode='gui', image_width=700, image_height=700)
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/simulator.py", line 86, in __init__
    self.load()
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/simulator.py", line 127, in load
    self.renderer = MeshRenderer(width=self.image_width,
  File "/gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py", line 91, in __init__
    logging.info(
Message: 'If you have trouble using EGL, please visit our trouble shooting guide'
Arguments: ('at http://svl.stanford.edu/igibson/docs/issues.html',)
ERROR: Invalid render_device choice: 0 < 0.
(igibson)[be244@c22n09 metagen_project_shared_space]$ cat 
^C
(igibson)[be244@c22n09 metagen_project_shared_space]$ vim /gpfs/loomis/project/jara-ettinger/be244/conda_envs/igibson/lib/python3.8/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py
(igibson)[be244@c22n09 metagen_project_shared_space]$ ldconfig -p | grep EGL
	libEGL_nvidia.so.0 (libc6,x86-64) => /lib64/libEGL_nvidia.so.0
	libEGL_mesa.so.0 (libc6,x86-64) => /lib64/libEGL_mesa.so.0
	libEGL_mesa.so.0 (libc6) => /lib/libEGL_mesa.so.0
	libEGL.so.1 (libc6,x86-64) => /lib64/libEGL.so.1
	libEGL.so.1 (libc6) => /lib/libEGL.so.1
	libEGL.so (libc6,x86-64) => /lib64/libEGL.so
`

ChengshuLi, 2021-03-16:
Can you provide a bit more information about your platform? (os, nvidia driver version, output of `nvidia-smi` etc)

Can you also show the output of `glxinfo`? Thanks!

fxia22, 2021-03-16:
also refer to #69. looks like these are similar issues.

betrevisan, 2021-03-17:
This is the output of glxinfo:

name of display: :1
display: :1  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0xffffffff)
    Device: llvmpipe (LLVM 7.0, 256 bits) (0xffffffff)
    Version: 18.3.4
    Accelerated: no
    Video memory: 128238MB
    Unified memory: no
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 7.0, 256 bits)
OpenGL version string: 2.1 Mesa 18.3.4
OpenGL shading language version string: 1.20
OpenGL extensions:
    GL_AMD_conservative_depth, GL_AMD_draw_buffers_blend, 
    GL_AMD_multi_draw_indirect, GL_AMD_seamless_cubemap_per_texture, 
    GL_AMD_shader_stencil_export, GL_AMD_shader_trinary_minmax, 
    GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
    GL_APPLE_packed_pixels, GL_ARB_ES2_compatibility, 
    GL_ARB_ES3_compatibility, GL_ARB_arrays_of_arrays, GL_ARB_base_instance, 
    GL_ARB_blend_func_extended, GL_ARB_buffer_storage, 
    GL_ARB_clear_buffer_object, GL_ARB_clear_texture, GL_ARB_clip_control, 
    GL_ARB_color_buffer_float, GL_ARB_compressed_texture_pixel_storage, 
    GL_ARB_conditional_render_inverted, GL_ARB_conservative_depth, 
[... 944 lines cut ...]
0x28a 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 16  0  0  0  0  0  0 0 None
0x28b 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x28c 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x28d 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x28e 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x28f 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x290 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x291 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x292 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x293 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x294 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x295 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x296 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x297 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x298 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x299 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x29a 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x29b 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x29c 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x29d 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x29e 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x29f 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x2a0 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None
0x2a1 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None
0x2a2 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None

betrevisan, 2021-03-17:
and this is the output of nvidia-smi:

Wed Mar 17 17:25:10 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.32.03    Driver Version: 460.32.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Tesla K80           On   | 00000000:83:00.0 Off |                    0 |
| N/A   33C    P8    25W / 149W |      0MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

betrevisan, 2021-03-17:
Sorry for the duplicated issue, the discussion on #69 already helps me a lot!

closed 2021-03-17 by betrevisan