StanfordVL/iGibson #69: ERROR: Invalid render_device choice: 0 < 0.

BurmanT, 2021-03-16:
Hi,

I was able to install gibson2 using pip install. However, when I try to run the demo using python -m gibson2.examples.demo.demo_static I keep getting an error that says ERROR: Invalid render_device choice: 0 < 0. This is how my terminal looks like. Any help would be appreciated! 

[image]
[image]

ChengshuLi, 2021-03-16:
Can you provide a bit more information about your platform? (os, nvidia driver version, output of `nvidia-smi` etc)

Can you also show the output of `glxinfo`? Thanks!

BurmanT, 2021-03-16:
Hello,

I am using a Linux cluster. I've attached the output of nvidia-smi and the beginning of the output of glxinfo. 

[image]
[image]
[image]

ChengshuLi, 2021-03-16:
Could you try one thing for me in the python terminal?
```
python

>>> import pybullet as p
>>> p.connect(p.GUI)
```
Can you confirm if you can see a pybullet GUI popping up?

BurmanT, 2021-03-16:
I am getting a Segmentation Fault 
[image]

fxia22, 2021-03-16:
Looks like this is using software rendering (mesa), instead of GPU accelerated rendering. The glxinfo should show vendor=Nvidia. You can refer to this trouble shooting guide for more help: 

http://svl.stanford.edu/igibson/docs/issues.html

ChengshuLi, 2021-04-13:
Is this issue resolved? Based on this similar issue: https://github.com/StanfordVL/iGibsonChallenge2021/issues/2, you might need to reinstall your nvidia driver.

closed 2021-06-30 by roberto-martinmartin

Atellas23, 2023-02-06:
Hey, I'm sorry I might have to reopen this issue😅
I'm using a Windows 11 machine, with an Nvidia GTX 1050 GPU and WSL2, Ubuntu 20.04. I intend to use iGibson with ROS, and so I was able to install iGibson with `pip install -e .` in the iGibson root following [this instructions](https://stanfordvl.github.io/iGibson/ros_integration.html). Whenever I try to run the minimal example though, it seems to fall apart. I have been fixing some stones in the way, but I don't seem to be able to solve `ERROR: Invalid render_device choice: 0 < 0`.

<details>
<summary>Output of <code>glxinfo -B</code></summary>
<br>

```
name of display: 192.168.178.102:0.0
display: 192.168.178.102:0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_get_drawable_type, GLX_EXT_import_context, 
    GLX_EXT_no_config_context, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_ATI_pixel_format_float, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_multithread_makecurrent, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_NV_float_buffer, 
    GLX_OML_swap_method, GLX_OML_sync_control, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce GTX 1050) (0xffffffff)
    Version: 21.2.6
    Accelerated: yes
    Video memory: 11078MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce GTX 1050)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 21.2.6
[... 445 lines cut ...]
0x261 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x262 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x266 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x267 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x268 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x269 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x26a 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x26b 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x26f 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x270 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x271 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x275 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x276 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x277 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x278 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x279 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x27a 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
```
</details>

In particular, you can see the `OpenGL version string` being `3.1 Mesa 21.2.6`, which I don't know how to fix anymore. I have (re)installed the drivers the way [Canonical tells developers to do so](htt [... cut ...] ons from [here](https://stanfordvl.github.io/iGibson/issues.html#unable-to-initialize-egl). I honestly don't know what else to do, I'm not an expert in any of these things, but I was hoping I could use iGibson as a realistic environment to generate data for a navigation project.

If any of you could help me or direct me to someone who could, that would be amazing! Thank you for taking the time to read this, and let me apologize preemtively just in case I did/said something really stupid. Cheers!

mjlbach, 2023-02-06:
Running iGibson in WSL is not supported. There are many remaining issues with opengl acceleration, cuda, nvidia, with WSL that we can't control.

Atellas23, 2023-02-06:
Oh, okay😞 Thanks for telling me!

mjlbach, 2023-02-06:
To clarify, these are generic WSL problems, not things we can work around. I would recommend either using iGibson on windows (which we support) or via dual-booting which should be a ~20 minute process.