StanfordVL/iGibson #70: Segmentation Fault using demo code on Mac OS

SamNPowers, 2021-03-22:
I followed the directions on http://svl.stanford.edu/igibson/docs/installation.html to install on my Mac (version 10.15.7). Specifically:

```
pip install gibson2
python -m gibson2.examples.demo.demo_static
```

Which results in this:
```
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets
INFO:root:Gibson Dataset path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/g_dataset
INFO:root:iG Dataset path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/cubicasa_dataset
INFO:root:Example path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/examples
INFO:root:Example config path: /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/examples/configs
pybullet build time: Mar 21 2021 14:02:23
WARNING:root:Rendering both iggui and pbgui is not supported on mac, choose either pbgui or iggui. Default to iggui.
INFO: Probing, EGL cannot run on this device
INFO:root:Using device 0 for rendering
INFO: Succesfully initialized GLFW context and window!
******************PyBullet Logging Information:
PyBullet Logging Information******************
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/mjcf_primitives/sphere8.obj
INFO:root:IndoorScene model: Rs
INFO:root:StaticIndoorScene scene: Rs
INFO:root:Loading traversable graph
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/g_dataset/Rs/mesh_z_up.obj
/Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:462: RuntimeWarning: divide by zero encountered in true_divide
  delta_uv1[:, 1] * delta_uv2[:, 0])
/Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:464: RuntimeWarning: invalid value encountered in multiply
  delta_pos2 * delta_uv1[:, 1][:, None]) * r[:, None]
/Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:466: RuntimeWarning: invalid value encountered in multiply
  delta_pos1 * delta_uv2[:, 0][:, None]) * r[:, None]
INFO:root:Loading robot model file: turtlebot/turtlebot.urdf
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/turtlebot/kobuki_description/meshes/main_body.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/turtlebot/kobuki_description/meshes/wheel.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_bottom.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/plate_bottom.obj
INFO:root:Loading /Users/powerss/.pyenv/versions/anaconda3-5.2.0/envs/venv_continual_rl4/lib/python3.7/site-packages/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_middle.obj
Segmentation fault: 11
```

This command fails in the same way:
```
python -m gibson2.examples.demo.env_example
```

As does using turtlebot_point_nav.yaml.

It seems the renderer is having some division by 0 issues? Also, pole_middle.obj always precedes a failure. Any pointers to fixing this? 

Thanks!

fxia22, 2021-03-24:
I cannot seem to reproduce this issue. Is this mac with integrated GPU or dedicated GPU? Can you confirm pybullet gui works on your platform? 

Could you try one thing for me in the python terminal?
```
python

>>> import pybullet as p
>>> p.connect(p.GUI)
```
Can you confirm if you can see a pybullet GUI popping up?

SamNPowers, 2021-03-24:
The GPU is a NVIDIA GeForce GT 650M. (The macbook is rather old, so that might be part of the issue - 2012 model.)

Pybullet GUI does indeed work, using `p.connect(p.GUI)`.

SamNPowers, 2021-03-25:
I just repro'd it again on a different Mac, this one a 2015 model running 11.2.3.

Here are my exact steps:
1. `conda create -n venv_gibson python=3.8`
2. `pip install gibson2`
3. `python -m gibson2.examples.demo.demo_static`

This time I retrieved this log:
```
Process:               python3.8 [4603]
Path:                  /Users/USER/*/python
Identifier:            python3.8
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        zsh [3109]
Responsible:           Terminal [3107]
User ID:               504

Date/Time:             2021-03-25 15:36:55.511 -0700
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Anonymous UUID:        CB62AC0E-E1ED-6B17-64DC-95BEAC15720F

Time Awake Since Boot: 4400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00007fd36c663002
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [4603]

VM Regions Near 0x7fd36c663002:
    MALLOC_LARGE             7fd36c600000-7fd36c663000 [  396K] rw-/rwx SM=PRV  
--> 
    MALLOC_LARGE             7fd36c674000-7fd36c6a6000 [  200K] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libGLImage.dylib              	0x00007fff6c9e3393 void glgConvertTo_32<GLGConverter_BGR8_ARGB8, (GLGMemory)2>(GLGOperationRec const*, GLDPixelModeRec const*) + 636
1   libGLImage.dylib              	0x00007fff6c9d6227 glgProcessPixelsWithProcessor + 3572
2   com.apple.AMDRadeonX4000GLDriver	0x0000000132394133 glrATIModifyTexSubImageCPU + 1394
3   com.apple.AMDRadeonX4000GLDriver	0x00000001323acf4c glrWriteTextureData + 588
4   GLEngine                      	0x00007fff6cbf99f7 glTexImage2D_Exec + 1323
5   libGL.dylib                   	0x00007fff6cbd9195 glTexImage2D + 55
6   GLFWRendererContext.cpython-38-darwin.so	0x000000012a7e389d MeshRendererContext::loadTexture(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float) + 493
7   GLFWRendererContext.cpython-38-darwin.so	0x000000012a7d1a02 void pybind11::cpp_function::initialize<pybind11::cpp_function::cpp_function<int, GLFWRendererContext, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float, pybind11::name, pybind11::is_method, pybind11::sibling, char [22]>(int (GLFWRendererContext::*)(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [22])::'lambda'(GLFWRendererContext*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float), int, GLFWRendererContext*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float, pybind11::name, pybind11::is_method, pybind11::sibling, char [22]>(int&&, GLFWRendererContext (*)(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, float), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [22])::'lambda'(pybind11::detail::function_call&)::operator()(pybind11::detail::function_call&) const + 338
8   GLFWRendererContext.cpython-38-darwin.so	0x000000012a7c1da5 pybind11::cpp_function::dispatcher(_object*, _object*, _object*) + 4261
9   python                        	0x000000010a8e3214 cfunction_call_varargs + 68
10  python                        	0x000000010a8e2467 _PyObject_MakeTpCall + 167

 ... [Truncated binary images for space] 

Model: MacBookPro11,5, BootROM 426.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.8 GHz, 16 GB, SMC 2.30f2
Graphics: kHW_AMDRadeonR9M370XItem, AMD Radeon R9 M370X, spdisplays_pcie_device, 2 GB
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1675.1)
Bluetooth: Version 8.0.3d9, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM1024G, 1 TB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: Microsoft® 2.4GHz Transceiver v8.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

ChengshuLi, 2021-04-13:
I tried to reproduce your issue but didn't manage to do it on my mac.

Could you try to monitor the cpu memory usage from Activity Monitor to check if the error is caused by out of memory?
If so, you could do the following steps to try:
- `git clone --recursive https://github.com/StanfordVL/iGibson.git`
- `cd iGibson && pip install -e .`

roberto-martinmartin, 2021-06-30:
Feel free to reopen the issue (or anyone with a similar problem). Thanks!

closed 2021-06-30 by roberto-martinmartin

kanishkg, 2021-09-16:
Was running into a similar issue. Disabling the `enable_shadows`parameter in the`MeshRenderSettings` seems to solve/bypass the issue.

shivanimall, 2021-09-20:
running on the latest ig facing the following issue when running `demo_static.py` on macOS

```
(ig) Nityas-Macbook-Pro:demo shivanimall$ python demo_static.py 
INFO:root:Importing iGibson (igibson module)
INFO:root:Assets path: /Users/shivanimall/iGibson/igibson/data/assets
INFO:root:Gibson Dataset path: /Users/shivanimall/iGibson/igibson/data/g_dataset
INFO:root:iG Dataset path: /Users/shivanimall/iGibson/igibson/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /Users/shivanimall/iGibson/igibson/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /Users/shivanimall/iGibson/igibson/data/cubicasa_dataset
INFO:root:iGibson Key path: /Users/shivanimall/iGibson/igibson/data/igibson.key
INFO:root:Example path: /Users/shivanimall/iGibson/igibson/examples
INFO:root:Example config path: /Users/shivanimall/iGibson/igibson/examples/configs
pybullet build time: Apr  1 2021 12:34:57
BDDL could not be imported - object taxonomy / abilities will be unavailable.
INFO: Probing, EGL cannot run on this device
INFO:root:Using device 0 for rendering
2021-09-20 18:39:41.904 Python[83908:45907625] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/fd/5jk_ll1j7gzbjtr57ltsdtww0000gn/T/org.python.python.savedState
INFO: Succesfully initialized GLFW context and window!
Traceback (most recent call last):
  File "/Users/shivanimall/iGibson/igibson/examples/demo/demo_static.py", line 33, in <module>
    DemoStatic().run_demo()
  File "/Users/shivanimall/iGibson/igibson/examples/demo/demo_static.py", line 19, in run_demo
    s = Simulator(mode="iggui", image_width=700, image_height=700)
  File "/Users/shivanimall/iGibson/igibson/simulator.py", line 161, in __init__
    self.load()
  File "/Users/shivanimall/iGibson/igibson/simulator.py", line 231, in load
    self.renderer = MeshRenderer(
  File "/Users/shivanimall/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 253, in __init__
    self.setup_framebuffer()
  File "/Users/shivanimall/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 327, in setup_framebuffer
    [
ValueError: not enough values to unpack (expected 9, got 8)

```