StanfordVL/iGibson #71: Second env fails to initialize when using multiprocessing

SamNPowers, 2021-03-24:
If I define some simple env like:
```
import os
from gibson2.envs.igibson_env import iGibsonEnv

class DiscreteGibsonEnv(iGibsonEnv):
    def __init__(self):
        config_path = <some config>
        super().__init__(config_file=config_path, scene_id="Rs_int", mode='headless')

```

and run:
```
import multiprocessing as mp

env_0 = DiscreteGibsonEnv()
env_proc = mp.Process(target=DiscreteGibsonEnv)
env_proc.start()
```

With the intention of creating one environment on the main thread, and the second in a process, the second environment seems to die during initialization, specifically during EglRenderContext.init(). It doesn't seem to matter if I run `.close()` on env_0 or not. 

Any suggestions? I'm running on Ubuntu 18.04 with CUDA 450.80.02. If I create both envs on the main thread, or both in processes, it seems to be fine. But one on each seems to break things.

Any ideas? Thanks!

fxia22, 2021-03-24:
Yes, this is a known issue. However, you can start multiple `iGibsonEnvs` in subprocesses, if there is no `iGibsonEnv` in the main process. Would this satisfy your use case? 

For your information, we also provide an example implementation of parallel environments: https://github.com/StanfordVL/iGibson/blob/master/gibson2/envs/parallel_env.py

SamNPowers, 2021-03-25:
Wow, thanks for the quick reply. 

I had noticed it mostly worked in multiple processes, as long as the main isn't involved - actually, part of the reason I asked is because I have two separate code paths. One of which was failing for this reason, and the other (which also starts one environment in the main process before spawning subprocesses that initialize their own envs) seems to work just fine. So far I haven't isolated what it is about the second that makes it work (despite some effort). I guess I was hoping it was a known issue with a known fix I had stumbled into on one code path and not the other.

I've also seen it happen with subprocesses only a couple times. Adding a couple second sleep before env construction seems to fix it, so far.

It's not especially a problem, now that I know what's happening. This issue meant I couldn't just drop the env into place in my infrastructure as I had been hoping. Perhaps the issue could be documented somewhere, at least? (Apologies if it already is and I missed it.)

(By the way, the environment/code base otherwise is great. Very easy to use and understand.)

roberto-martinmartin, 2021-06-30:
Closing this. Feel free to reopen if there is something to be improved

closed 2021-06-30 by roberto-martinmartin