StanfordVL/iGibson #72: Where to access person meshes dataset? It seems not available in the official documentation.

mustang66ytz, 2021-03-31:


ChengshuLi, 2021-03-31:
@fxia22 Would you mind putting the person meshes in the assets?

fxia22, 2021-04-01:
added, you can run `wget https://storage.googleapis.com/gibson_scenes/assets_igibson.tar.gz` to update the assets.

Note that the person meshes are also included in the data from the cvpr challenge repo, which you can download here: https://github.com/StanfordVL/iGibsonChallenge2021/blob/master/download.sh

mustang66ytz, 2021-04-01:
I wander is there any specific scene instance or example with pedestrians moving around following orca rules with simulator running simultaneously?

ChengshuLi, 2021-04-01:
Hi, are you trying to participate in the [iGibson Challenge 2021](http://svl.stanford.edu/igibson/challenge.html)? If so, you can follow our [started code](https://github.com/StanfordVL/iGibsonChallenge2021). 
If not, I can point you to the [social nav task](https://github.com/StanfordVL/iGibson/blob/cvpr21_challenge/gibson2/tasks/social_nav_random_task.py) that incorporates orca and the corresponding [config file](https://github.com/StanfordVL/iGibson/blob/cvpr21_challenge/gibson2/examples/configs/locobot_social_nav.yaml). 

You can try
```
cd gibson2/envs
python igibson_env.py --config ../../examples/configs/locobot_social_nav.yaml --mode gui
```

mustang66ytz, 2021-04-02:
Cool! That is awesome!

ChengshuLi, 2021-04-13:
Feel free to re-open the issue if you have further questions!

closed 2021-04-13 by ChengshuLi