StanfordVL/iGibson #73: Object Demo doesn't show textures

yashkant, 2021-04-03:
Hi,

Thanks for the code! 

I ran the [objects demo](http://svl.stanford.edu/igibson/docs/objects.html#examples) from this [file](https://github.com/StanfordVL/iGibson/blob/master/gibson2/examples/demo/object_example.py), but I do not see the object textures being loaded, see the image below. 

What might have gone wrong? 

[image]

roberto-martinmartin, 2021-04-03:
Hi,

You are visualizing with pybullet viewer. Try iGibsons viewer to see the result of iGibson rendering. Let us know if that works.

yashkant, 2021-04-03:
Thanks for the quick reply and the suggestion, I'll try this and would reopen if I am unable to figure this out.

closed 2021-04-03 by yashkant

fxia22, 2021-04-03:
Hi @yashkant , we just updated the assets folder, and if you delete the `gibson2/data/assets` folder and run `python -m gibson2.utils.assets_utils --download_assets` again, you should be able to get these models with textures.