StanfordVL/iGibson #74: determine mesh face & vertex in obj file for collision points

shivani1494, 2021-04-05:
in iGibson,  is there a function/way to determine which triangular mesh face (in obj file) does the cartesian world coordinate for the collision point returned on the object from `getContactPoints` func (pybullet) belong to? 

knowing which face of the triangular mesh object will allow me determine the vertexes associated with that face, which is what I am interested in.

Let me know if my question is unclear. Thanks!

fxia22, 2021-04-06:
Hi @shivani1494 , I believe `getContactPoints` only returns the contact points in world coordinates, but there is no information about which face / vertices are associated with the contact points. 

However, it is not too difficult to get this information yourself, there is an undocumented API called p.getMeshData which dumps mesh information of an object, you can find it [here](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#) by searching for `getMeshData`. You will need to iterate over the faces and test if the contact points are in / close to a certain face. To make this process faster, you can also leverage something like BVH or KDTree.

ChengshuLi, 2021-04-13:
Feel free to re-open the issue if you have further questions!

closed 2021-04-13 by ChengshuLi