StanfordVL/iGibson #75: motion planning demo failed

sunniee, 2021-04-15:
Hi, I used docker `igibson/igibson-gui:latest` to run your example. I tried motion planning demo using two demo scripts:

First script: `python motion_planning_example.py -c ../configs/turtlebot_interactive_nav.yaml`.
Error is 
`Traceback (most recent call last):`
`  File "motion_planning_example.py", line 32, in <module>`
`    run_example(args)`
`  File "motion_planning_example.py", line 12, in run_example`
`    motion_planner = MotionPlanningWrapper(nav_env)`
`  File "/opt/igibson_new/gibson2/utils/motion_planning_wrapper.py", line 48, in __init__`
`    assert 'occupancy_grid' in self.env.output`

Second script: `python mouse_interaction.py` and click `m` to change to `planning` mode. Every time I `Left click`, error occurs.
Error is 
`Traceback (most recent call last):`
  `File "/opt/igibson_new/gibson2/render/viewer.py", line 423, in mouse_callback`
    `self.planner.set_marker_position_yaw(`
`AttributeError: 'NoneType' object has no attribute 'set_marker_position_yaw'`
I think the variable of the self.planner is not initialized

Can you help me to fix these two issues? Or are there other ways to run motion planning example?

shrirangsp, 2021-04-15:
Hi, 
I had same issue. For the first one you need to add occupancy_grid [](https://github.com/StanfordVL/iGibson/blob/0f63ece61f3a6113f6842b71acc63fccbc56f02d/gibson2/examples/configs/turtlebot_interactive_nav.yaml#L47) here.

ChengshuLi, 2021-04-15:
The command to run is:
```
python motion_planning_example.py -m iggui -c ../configs/fetch_room_rearrangement.yaml
```
You would need to add `occupancy_grid` to `output` in `fetch_room_rearrangement.yaml`, like @shrirangsp has suggested.

The second script `python mouse_interaction.py` does not support planning. Please use the script above.

Please let us know if this resolves your issue. Thank you!

ChengshuLi, 2021-04-15:
This [commit](https://github.com/StanfordVL/iGibson/commit/28336ab41a7851d652c8dc17b1d42bc27757a553) should make the process even easier. 

You only need to run `python motion_planning_example.py` now.

sunniee, 2021-04-16:
@ChengshuLi @shrirangsp Thanks! It ran successfully. But after I `Left Click` one point, the robot didn't do the planning or move to the point I selected. Just like the picture below. Are there any other operations needed to do?
[image]

sunniee, 2021-04-16:
The Introduction in documents is like that:

> Left click and drag: create (click), visualize (drag) and plan / execute (release) a base motion subgoal for the robot base to reach the physical point that corresponds to the clicked pixel

After I do the `Left click`, there is no error, except for the arrow, nothing happens ...

fxia22, 2021-04-16:
It is because your motion planning goal is too far away from the robot. The motion planner relies on a local occupancy map built by LiDAR, so you can try setting a goal in front of the robot within a few meters. We don't have a global motion planner at this moment.

sunniee, 2021-04-17:
@fxia22 It works! Thank you!

ChengshuLi, 2021-04-18:
Feel free to re-open the issue if you have further questions!

closed 2021-04-18 by ChengshuLi