StanfordVL/iGibson #76: Object Rendering

shrirangsp, 2021-04-15:
Hi,

Setup: Im trying to render a Cracker box object using custom camera setup.

Current Behaviour: The external view does not show the object while the Camera Setup window shows the Cracker Box as black.
[image]

`def main():

    #Initialize
    config_filename = os.path.join(gibson2.example_config_path, 'tiago_reaching_mod.yaml')
    env = iGibsonEnv(config_file=config_filename, mode='gui')
    envSim = env.simulator
    #renderer = MeshRenderer(width=512, height=512)
    settings = MeshRendererSettings(enable_shadow=False, msaa=False, enable_pbr=False)

    #Robot
    robot = env.robots[0]
    robot.robot_body.reset_position([0,-1.2,0])
    robot.robot_body.reset_orientation([0,0,90,-90])

    # Load Object
    cabinet_0007 = os.path.join(gibson2.assets_path, 'models/cabinet2/cabinet_0007.urdf')
    obj1 = ArticulatedObject(filename=cabinet_0007)
    envSim.import_object(obj1)
    obj1.set_position([1, 0, 0])
    obj1.set_position_orientation([0, 0, 0], [0, 0, 0, 90])

    obj2 = YCBObject('003_cracker_box')
    envSim.import_object(obj2)
    obj2.set_position_orientation([0, 0, 0.4], [0, 0, 90, 90])

    # #Render Object Using PyBullet
    time.sleep(20)

    env.simulator.renderer.set_camera(camera_pose, camera_pose+view_direction, up)
    env.simulator.renderer.setup_framebuffer()
    env.simulator.renderer.set_fov(45)
    frame = cv2.cvtColor(np.concatenate(envSim.renderer.render(modes=('rgb')), axis=1),
                         cv2.COLOR_RGB2BGR)

    cv2.imshow('Camera Setup', frame)`

ChengshuLi, 2021-04-15:
I tried to reproduce this issue and I found that the position of the YCB object is too low and it's already contained in the cabinet. You can try something like `obj2.set_position_orientation([0, 0, 0.65], [0, 0, 90, 90])`. 
The screenshot is attached with the original position `[0, 0, 0.4]`.
[image]

I was not able to reproduce the black texture. Have you tried the first option in #56: Download the full iG dataset, by running `python -m gibson2.utils.assets_utils --download_ig_dataset`?

Thanks!

fxia22, 2021-04-15:
One thing to pay special attention to is that if the environment texture (in ig_dataset) is loaded, there is a warning in the command line output if environment texture cannot be found.

shrirangsp, 2021-04-15:
> I tried to reproduce this issue and I found that the position of the YCB object is too low and it's already contained in the cabinet. You can try something like `obj2.set_position_orientation([0, 0, [...]

Oh thanks! Now I can see the object in the External View but still see a black box in Camera Setup Window

ChengshuLi, 2021-04-18:
Please feel free to try the suggestions that Fei and I listed above and let us know how it goes.

shrirangsp, 2021-04-19:
I have tried your suggestion but still I see black box in my Camera setup window. 
What I want to do is, I need to get the rgb image and depth data by placing camera directly on top of the object. Please help me in solving the issue.

ChengshuLi, 2021-04-19:
Please try to remove `env.simulator.renderer.setup_framebuffer()`. 
Also, you would need to call `env.simulator.sync()` after importing the cracker box and cabinet.

You can move the camera location by `env.simulator.renderer.set_camera(camera_pose, camera_pose+view_direction, up)` and adjust the fov by `env.simulator.renderer.set_fov(45)`. You can specify the image width and height in the config file.

Here is the full code:
```
from gibson2.envs.igibson_env import iGibsonEnv
import gibson2
import os
import numpy as np
import cv2

from gibson2.objects.articulated_object import ArticulatedObject
from gibson2.objects.ycb_object import YCBObject

# Initialize
config_filename = os.path.join(
    gibson2.example_config_path, 'turtlebot_point_nav.yaml')
env = iGibsonEnv(config_file=config_filename, mode='headless')

# Load Object
cabinet_0007 = os.path.join(
    gibson2.assets_path, 'models/cabinet2/cabinet_0007.urdf')
obj1 = ArticulatedObject(filename=cabinet_0007)
env.simulator.import_object(obj1)
obj1.set_position([1, 0, 0])
obj1.set_position_orientation([0, 0, 0], [0, 0, 0, 90])

obj2 = YCBObject('003_cracker_box')
env.simulator.import_object(obj2)
obj2.set_position_orientation([0, 0, 0.65], [0, 0, 90, 90])

env.simulator.sync()

camera_pose = np.array([-1.5, 0, 1.2])
view_direction = np.array([0.9, 0, -0.16])
up = np.array([0, 0, 1])

env.simulator.renderer.set_camera(camera_pose, camera_pose+view_direction, up)
frame = cv2.cvtColor(
    np.concatenate(env.simulator.renderer.render(modes=('rgb')), axis=1),
    cv2.COLOR_RGB2BGR)
cv2.imwrite('test.png', (frame * 255).astype(np.uint8))
```

The output image should look like this:
[image]

shrirangsp, 2021-04-19:
Thank you @ChengshuLi ! env.simulator.sync() solved the problem.

closed 2021-04-19 by shrirangsp