StanfordVL/iGibson #79: Cannot download the dataset from Gibson Database of 3D Spaces

jjanixe, 2021-04-21:
Hi @fxia22,
I tried to download the Gibson2 Room Dataset from https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform,
and I couldn't access the cloud storage because of the following issue.

This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>UserProjectAccountProblem</Code>
<Message>User project billing account not in good standing.</Message>
<Details>The billing account for the owning project is disabled in state absent</Details>
</Error>

Could you please check if the payment was properly made?

fxia22, 2021-04-21:
Will fix this asap, to understand more, which `tar.gz` file are you trying to download?

jjanixe, 2021-04-21:
Thanks for the quick reply!
I was trying to download
All scenes, 572 scenes (108GB): gibson_v2_all.tar.gz  to use with iGibson(2020), 
Gibson Standard Navigation Benchmark, 
and Dataset Splits.

fxia22, 2021-04-21:
Ok, sure, I will use another cloud bucket. Please give us some time to re-upload the data. thanks!

fxia22, 2021-04-21:
@jjanixe the new links are up, while the upload will finish in the next couple of hours.

jjanixe, 2021-04-21:
Thank you :) I'll close the issue if the download succeeds from the new links.

jjanixe, 2021-04-21:
Sorry, but I just found out that the same issue occurred for the links provided for Gibson Standard Navigation Benchmark and Dataset Splits still do not work.
Could you please check if the correct links were uploaded?

fxia22, 2021-04-21:
you can use https://storage.googleapis.com/gibson_scenes/splits.tar.gz

fxia22, 2021-04-21:
The splits files are uploaded to https://storage.googleapis.com/gibson_scenes/splits.tar.gz while we are trying to recover the scenario files.

fxia22, 2021-04-21:
@jjanixe updated, pls try again, thx!

closed 2021-04-22 by jjanixe

aqiu2018, 2021-11-01:
@fxia22 hi, I'am facing the same problem,  how can I fix it? 

<Error>
<Code>UserProjectAccountProblem</Code>
<Message>User project billing account not in good standing.</Message>
<Details>The billing account for the owning project is disabled in state absent</Details>
</Error>

fxia22, 2021-11-03:
@aqiu2018 we are in the process of switching billing accounts and will fix it soon.

fxia22, 2021-11-03:
@aqiu2018 access restored, let us know if you still have issues.