StanfordVL/iGibson #81: Black screen for mesh_renderer_gpu_example.py and mesh_renderer_simple_example.py

safoex, 2021-04-21:
After clean install from pip I have a black screen instead of (supposed) rgb image in both scripts:

mesh_renderer_gpu_example.py
[image]

mesh_renderer_simple_example.py
[image]

output of e.g. mesh_renderer_gpu_example.py:

```
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/assets
INFO:root:Gibson Dataset path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/cubicasa_dataset
INFO:root:Example path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/examples
INFO:root:Example config path: /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/examples/configs
pybullet build time: Apr 16 2021 03:37:45
INFO:root:Device 0 is available for rendering
INFO:root:Command '['/home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/render/mesh_renderer/build/test_device', '2']' returned non-zero exit status 1.
INFO:root:Device 2 is not available for rendering
INFO:root:Using device 0 for rendering
WARNING:root:Environment texture not available, cannot use PBR.
INFO:root:Using cuda device 0 for pytorch
INFO:root:Loading /home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/data/g_dataset/Rs/mesh_z_up.obj
/home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:462: RuntimeWarning: divide by zero encountered in true_divide
  delta_uv1[:, 1] * delta_uv2[:, 0])
/home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:464: RuntimeWarning: invalid value encountered in multiply
  delta_pos2 * delta_uv1[:, 1][:, None]) * r[:, None]
/home/safoex/Documents/code/objectrenderer/igibson/lib/python3.6/site-packages/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:466: RuntimeWarning: invalid value encountered in multiply
  delta_pos1 * delta_uv2[:, 0][:, None]) * r[:, None]
[Object(0)->VAO([0])] [Instance(0) -> Object(0)]
{0: Material(material_type: texture, texture_id: 9, metallic_texture_id:None, roughness_texture_id:None, normal_texture_id:None, color: [0.5, 0.5, 0.5]), 1: Material(material_type: color, texture_id: -1, metallic_texture_id:None, roughness_texture_id:None, normal_texture_id:None, color: [0.5, 0.5, 0.5])} [0]
Render: 71.75 fps, 0.01394 seconds
[many lines of profiler output]

```

fxia22, 2021-04-21:
looks like `WARNING:root:Environment texture not available, cannot use PBR.`, can you download the ig_dataset, since the environment textures are there? Or turn off PBR? (by setting `enable_pbr=False` when creating `MeshRendererSettings`. thanks!

safoex, 2021-04-22:
Many thanks, `enable_pbr=False` helped, now I have rgb image as well!

However, when I want to render some other object, e.g. one YCB object, I have both screens black :( 
I try to substitute `model_path` with some YCB object path as follows:
```
model_path = YCBObject('003_cracker_box').visual_filename
# model_path = os.path.join(get_scene_path('Rs'), 'mesh_z_up.obj')
```
I checked that YCB object models and textures loaded correctly to this path...
How to correctly load YCB objects ?

ChengshuLi, 2021-04-26:
The solution is to change the camera pose and viewing direction so that the camera points at the YCB object, which is at the location [0, 0, 0].

You can change these lines in `mesh_renderer_gpu_example.py` to
```
     camera_pose = np.array([0.5, 0.5, 0.5])
     view_direction = np.array([-0.5, -0.5, -0.5])
```
and run
```
python mesh_renderer_gpu_example.py ../../data/assets/models/ycb/003_cracker_box/textured_simple.obj
```

You should see something like this
[image]

I am running with `enable_pbr=True`. You can download the environment texture (which is needed for PBR, as suggested by Fei above) using the following script:
```
python -m gibson2.utils.assets_utils --download_ig_dataset
```

safoex, 2021-04-27:
I changed camera pose / view direction and it worked out!

Even better for my case, if I disable pbr by `enable_pbr=False`, I have YCB object on a black background:
[image]

Thanks for prompt reply!! I can close the issue now.

closed 2021-04-27 by safoex