StanfordVL/iGibson #85: Installation error

shrirangsp, 2021-04-28:
Hello,

Im trying to install on a fresh installation of Ubuntu 16.04 with ROS Kinetic

When i run pip install -e .

I get following errors:

`Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-Ua7TWx/scipy/setup.py", line 31, in <module>
        raise RuntimeError("Python version >= 3.7 required.")
    RuntimeError: Python version >= 3.7 required.
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-Ua7TWx/scipy/

Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-h4ymFj/opencv-python/setup.py", line 10, in <module>
        import skbuild
    ImportError: No module named skbuild
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-h4ymFj/opencv-python/

`

fxia22, 2021-05-19:
It seems the problem is coming from `scipy` and `opencv-python` instead of iGibson.

roberto-martinmartin, 2021-06-30:
Could you provide more information? Did you solve the opencv-python issue?
Feel free to reopen if it is a problem with Gibson.
Thanks!

closed 2021-06-30 by roberto-martinmartin