StanfordVL/iGibson #87: how does nav_env.test_valid_position() work?

yimengli46, 2021-05-06:
I am trying to densely render observations at multiple free positions of the environment.
Sometimes when return value of `nav_env.test_valid_position()` is True, the observation is totally black, indicating current position is not free.

I also observed that if we use floor_trav_0.png as an occupancy map of an interactive scene, some furniture like bed is also traversable.

So how to know if the position is free?

yimengli46, 2021-05-07:
And I got some very different rendering results.
This one is correct,
[image]
These two are weird. It seems to me there is larger environment outside of the scene.
[image]
[image]

I wonder why.

sycz00, 2021-05-08:
Hey, 

the thing with the larger environment happens to me as well. I use "Benevolence_1_int" and outside the doors looks similarly blurry as in your picture.

Sorry for just a comment and no answer :(

yimengli46, 2021-05-08:
I think I figured it out. 
The trav_map_resolution is 0.01 rather than 0.1 as mentioned in the config file.

Using the trav_map as reference to sample the observations would work.
I

closed 2021-05-12 by yimengli46