StanfordVL/iGibson #91: Pybullet not responding

zxwang96, 2021-05-20:
Hi, I tried to run the example script `iGibson/gibson2/examples/demo/motion_planning_example.py`. I run it with options `-c examples/configs/locobot_social_nav.yaml -m gui`. After that the terminal output info normally like this:
`INFO:root:/home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/window_57_0.urdf
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/walls_0.urdf
INFO:root:Moving URDF to (pos,ori): [0. 0. 0.], [0. 0. 0. 1.]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/floors_0.urdf
INFO:root:Moving URDF to (pos,ori): [0. 0. 0.], [0. 0. 0. 1.]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/ceilings_0.urdf
INFO:root:Moving URDF to (pos,ori): [0. 0. 0.], [0. 0. 0. 1.]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/bottom_cabinet_0_0.urdf
INFO:root:Moving URDF to (pos,ori): [-1.78945398 -0.50183213  0.36946967], [0.         0.         0.71240277 0.70177083]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/plant_1_0.urdf
INFO:root:Moving URDF to (pos,ori): [-1.80831023 -0.51558935  1.0704797 ], [0.         0.         0.71148495 0.70270134]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/sofa_2_0.urdf
INFO:root:Moving URDF to (pos,ori): [-1.6146312  -1.70583687  0.52357131], [0.         0.         0.7078658  0.70634695]
INFO:root:Loading /home/wzx/workspace/iGibson/gibson2/data/ig_dataset/scene_instances/20210520-003019_12969329269456702467_6307/floor_lamp_3_0.urdf
INFO:root:Moving URDF to (pos,ori): [-1.84866646  0.01634411  0.91635253], [0. 0. 0. 1.]
`
But the robot_viewer, pybullet window were just not responding. Can you help me with this issue?  Thank you!

ChengshuLi, 2021-09-17:
@zxwang96 You would need to use your mouse to click on the ExternalView to send commands to the robot.
Related issue: https://github.com/StanfordVL/iGibson/issues/75

fxia22, 2021-09-22:
Feel free to re-open the issue if you have further questions.

closed 2021-09-22 by fxia22