StanfordVL/iGibson #96: docker vnc not showing gui

WoojuLee24, 2021-06-29:
I installed igibson/igibson-gui:latest and igibson/igibson:latest according to http://svl.stanford.edu/igibson/docs/installation.html#installing-the-environment

When I execute 
cd iGibson
./docker/headless-gui/run.sh
# run a GUI example after the container command line prompt shows:
python simulator_example.py

PyBullet does not visualize anything. 
How can I simulate the environment?
This is the the result.

./docker/headless-gui/run.sh
Starting VNC server on port 5900 with password 112358
please run "python simulator_example.py" once you see the docker command prompt:
root@93c40d5ab73e:/opt/igibson/gibson2/examples/demo# python simulator_example.py 
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /opt/igibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /opt/igibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /opt/igibson/gibson2/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /opt/igibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /opt/igibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /opt/igibson/gibson2/examples
INFO:root:Example config path: /opt/igibson/gibson2/examples/configs
pybullet build time: Dec 23 2020 01:46:51
INFO:root:Device 0 is available for rendering
INFO:root:Using device 0 for rendering
WARNING:root:Environment texture not available, cannot use PBR.
******************PyBullet Logging Information:
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 10.0.0, 256 bits)
GL_VERSION=3.3 (Core Profile) Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=3.30
pthread_getconcurrency()=0
Version = 3.3 (Core Profile) Mesa 20.0.8
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 10.0.0, 256 bits)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
PyBullet Logging Information******************
ven = VMware, Inc.
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/mjcf_primitives/sphere8.obj
INFO:root:IndoorScene model: Rs
INFO:root:StaticIndoorScene scene: Rs
INFO:root:Loading traversable graph
INFO:root:Loading /opt/igibson/gibson2/data/g_dataset/Rs/mesh_z_up.obj
/opt/igibson/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:462: RuntimeWarning: divide by zero encountered in true_divide
  delta_uv1[:, 1] * delta_uv2[:, 0])
/opt/igibson/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:464: RuntimeWarning: invalid value encountered in multiply
  delta_pos2 * delta_uv1[:, 1][:, None]) * r[:, None]
/opt/igibson/gibson2/render/mesh_renderer/mesh_renderer_cpu.py:466: RuntimeWarning: invalid value encountered in multiply
  delta_pos1 * delta_uv2[:, 0][:, None]) * r[:, None]
INFO:root:Loading robot model file: turtlebot/turtlebot.urdf
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/kobuki_description/meshes/main_body.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/kobuki_description/meshes/wheel.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_bottom.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/plate_bottom.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_middle.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/plate_middle.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_top.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/pole_kinect.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/stacks/hexagons/plate_top.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/turtlebot/turtlebot_description/meshes/sensors/kinect.obj
INFO:root:Loading /opt/igibson/gibson2/data/assets/models/ycb/003_cracker_box/textured_simple.obj
[Instance(0) -> Object(0), Instance(1) -> Object(1), Instance(2) -> Object(2), Robot(3) -> Objects(3,4,5,5,6,7,8,8,8,8,8,8,9,10,10,10,10,11,12,12,12,12,13,13,14,15), Instance(4) -> Object(16), Instance(5) -> Object(16), Instance(6) -> Object(16), Instance(7) -> Object(16), Instance(8) -> Object(16), Instance(9) -> Object(16), Instance(10) -> Object(16), Instance(11) -> Object(16), Instance(12) -> Object(16), Instance(13) -> Object(16)]
ven = VMware, Inc.
Simulator step: 16.53 fps, 0.06051 seconds
Simulator step: 17.01 fps, 0.05878 seconds
Simulator step: 19.50 fps, 0.05127 seconds
Simulator step: 21.92 fps, 0.04562 seconds
Simulator step: 20.25 fps, 0.04937 seconds
Simulator step: 37.57 fps, 0.02662 seconds
Simulator step: 55.67 fps, 0.01796 seconds

fxia22, 2021-06-29:
You would need to connect to this docker image via vnc, you can connect to `127.0.0.1:5900` with vnc viewer software.

closed 2021-06-30 by roberto-martinmartin

BIT-TYJ, 2024-04-12:
Hello, I run the docker on the server, while the server is display via vnc, how do I visualise docker?