StanfordVL/iGibson #101: ImportError: cannot import name 'tinyobjloader'

Tridolly, 2021-07-05:
when I run `python -m gibson2.examples.demo.env_example` according to the official tutorial, I got this error:

```
INFO:root:Importing iGibson (gibson2 module)
INFO:root:Assets path: /home/we/textModel/iGibson/gibson2/data/assets
INFO:root:Gibson Dataset path: /home/we/textModel/iGibson/gibson2/data/g_dataset
INFO:root:iG Dataset path: /home/we/textModel/iGibson/gibson2/data/ig_dataset
INFO:root:3D-FRONT Dataset path: /home/we/textModel/iGibson/gibson2/data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: /home/we/textModel/iGibson/gibson2/data/cubicasa_dataset
INFO:root:Example path: /home/we/textModel/iGibson/gibson2/examples
INFO:root:Example config path: /home/we/textModel/iGibson/gibson2/examples/configs
pybullet build time: Jul  1 2021 15:32:31
Traceback (most recent call last):
  File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/we/textModel/iGibson/gibson2/examples/demo/env_example.py", line 1, in <module>
    from gibson2.envs.igibson_env import iGibsonEnv
  File "/home/we/textModel/iGibson/gibson2/envs/igibson_env.py", line 2, in <module>
    from gibson2.envs.env_base import BaseEnv
  File "/home/we/textModel/iGibson/gibson2/envs/env_base.py", line 10, in <module>
    from gibson2.simulator import Simulator
  File "/home/we/textModel/iGibson/gibson2/simulator.py", line 4, in <module>
    from gibson2.render.mesh_renderer.mesh_renderer_cpu import MeshRenderer
  File "/home/we/textModel/iGibson/gibson2/render/mesh_renderer/mesh_renderer_cpu.py", line 3, in <module>
    from gibson2.render.mesh_renderer import tinyobjloader
ImportError: cannot import name 'tinyobjloader'
```

I am new to ibision and hope to get answer

fxia22, 2021-07-05:
This is caused by the fact that installation is not correct. Did you run `git clone --recursive <repo_url>` and `pip install -e .`?

Tridolly, 2021-07-05:
> This is caused by the fact that installation is not correct. Did you run `git clone --recursive <repo_url>` and `pip install -e .`?

yes, I've run these two commands before, but got an error.

```
Installing collected packages: gibson2
  Attempting uninstall: gibson2
    Found existing installation: gibson2 1.0.1
    Uninstalling gibson2-1.0.1:
      Successfully uninstalled gibson2-1.0.1
  Running setup.py develop for gibson2
    ERROR: Command errored out with exit status 1:
     command: /home/we/anaconda3/envs/py3-igibson/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/we/textModel/iGibson/setup.py'"'"'; __file__='"'"'/home/we/textModel/iGibson/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /home/we/textModel/iGibson/
    Complete output (65 lines):
    running develop
    running egg_info
    writing gibson2.egg-info/PKG-INFO
    writing dependency_links to gibson2.egg-info/dependency_links.txt
    writing requirements to gibson2.egg-info/requires.txt
    writing top-level names to gibson2.egg-info/top_level.txt
    reading manifest file 'gibson2.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'gibson2.egg-info/SOURCES.txt'
    running build_ext
    -- The C compiler identification is GNU 7.5.0
    -- The CXX compiler identification is GNU 7.5.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    -- pybind11 v2.6.0 dev
    -- Found PythonInterp: /home/we/anaconda3/envs/py3-igibson/bin/python (found version "3.6.10")
    -- Found PythonLibs: /home/we/anaconda3/envs/py3-igibson/lib/libpython3.6m.so
    -- Performing Test HAS_FLTO
    -- Performing Test HAS_FLTO - Success
    CMake Error at /usr/share/cmake-3.10/Modules/FindCUDA.cmake:682 (message):
      Specify CUDA_TOOLKIT_ROOT_DIR
    Call Stack (most recent call first):
      CMakeLists.txt:45 (find_package)
    
    
    -- Configuring incomplete, errors occurred!
    See also "/home/we/textModel/iGibson/build/temp.linux-x86_64-3.6/CMakeFiles/CMakeOutput.log".
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/we/textModel/iGibson/setup.py", line 156, in <module>
        include_package_data=True,
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/site-packages/setuptools/__init__.py", line 165, in setup
        return distutils.core.setup(**attrs)
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/site-packages/setuptools/command/develop.py", line 38, in run
        self.install_for_development()
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/site-packages/setuptools/command/develop.py", line 140, in install_for_development
        self.run_command('build_ext')
      File "/home/we/anaconda3/envs/py3-igibson/lib/python3.6/distutils/cmd.py", line 313, in run_command
[... 33 lines cut ...]
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~hysics
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/render/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~ender
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/reward_functions/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~eward_functions
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/robots/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~obots
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/scenes/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~cenes
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/sensors/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~ensors
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/simulator.py
   from /tmp/pip-uninstall-5pwsucj6/simulator.py
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/tasks/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~asks
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/termination_conditions/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~ermination_conditions
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/test/
   from /home/we/.local/lib/python3.6/site-packages/gibson2/~est
  Moving to /home/we/.local/lib/python3.6/site-packages/gibson2/utils/
[... cut ...]
ERROR: Command errored out with exit status 1: /home/we/anaconda3/envs/py3-igibson/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/we/textModel/iGibson/setup.py'"'"'; __file__='"'"'/home/we/textModel/iGibson/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.
```

I have no idea about this at all

fxia22, 2021-07-05:
looks like you don't have CUDA installed, you can either install cuda, or set `USE_CUDA` to False here: 
https://github.com/StanfordVL/iGibson/blob/master/gibson2/render/CMakeLists.txt#L11

Tridolly, 2021-07-06:
> looks like you don't have CUDA installed, you can either install cuda, or set `USE_CUDA` to False here:
> https://github.com/StanfordVL/iGibson/blob/master/gibson2/render/CMakeLists.txt#L11

After installing CUDA, I successfully solved this problem. thx ;-)

closed 2021-07-06 by Tridolly

Nelx-SJTU, 2022-04-10:
> looks like you don't have CUDA installed, you can either install cuda, or set `USE_CUDA` to False here: https://github.com/StanfordVL/iGibson/blob/master/gibson2/render/CMakeLists.txt#L11

I think the codes were modified, where should I set USE_CUDA false now ?