StanfordVL/iGibson #102: Exception: floor_heights.txt cannot be found in mode: area1

AaronWaxson, 2021-07-06:
I downloaded the 2d3ds_for_igibson.zip latest version, and solve the 'floors.txt' exception.
But  When I run `roslaunch gibsion-ros turtlebot_rgbd.launch` I get a new one that is:
Exception: floor_heights.txt cannot be found in mode: area1
[image]

Any idea about that ? I run in Ubuntu 16, python2.7.

And I downloaded gibson_fullplus dataset, and after I unzipped it. I didn't see the dataset format like document told. I only have that:
[image]
 Is there any problem? Help plz.

fxia22, 2021-08-06:
@AaronWaxson you will need to download this dataset: 2d3ds_for_gibson.tar.gz

You can get the link here: https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform

AaronWaxson, 2021-08-09:
> @AaronWaxson you will need to download this dataset: 2d3ds_for_gibson.tar.gz
> 
> You can get the link here: https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform

I always downloaded datasets in this link, and I downloaded them again and again, but I still got this:
[image]
And I check every folder, there's no floor_heights.txt file like this:

[image]

Is that right ? And I still could not launch ros example files.
What should I do to fix this problem ? I has been broken down! Please help!

elhamAm, 2021-08-17:
I have the same issue with the Matterport dataset. This problem exits for a lot of the folders in the gibson dataset such as Hitchland.

Exception: floor_heights.txt cannot be found in model: house1

ChengshuLi, 2021-09-17:
@AaronWaxson 
Could you please follow the new documentation we added here? http://svl.stanford.edu/igibson/docs/dataset.html#download-gibson-and-stanford-2d-3d-semantics-scenes

Please download `2d3ds_for_igibson.zip`. Thanks!

ChengshuLi, 2021-09-17:
@elhamAm I responded to you in the other issue (https://github.com/StanfordVL/iGibson/issues/128). We have updated our documentation. Sorry for the inconvenience.

ChengshuLi, 2021-09-22:
Please feel free to re-open the issue if the problem persists. Thanks!