StanfordVL/iGibson #108: Can not run with ROS

DavidYaonanZhu, 2021-07-29:
I followed the tutorial to run Igibson ros integration, installation works fine

but when I run

```
roslaunch igibson-ros turtlebot_rgbd.launch
```
the error is AssertionError: Scene area1 does not exist

DavidYaonanZhu, 2021-07-29:
I checked another issues describing the matter, I downloaded data files, should I modify some file path?

fxia22, 2021-08-06:
@DavidYaonanZhu you will need to download this dataset: 2d3ds_for_gibson.tar.gz

You can get the link here: https://docs.google.com/forms/d/e/1FAIpQLScWlx5Z1DM1M-wTSXaa6zV8lTFkPmTHW1LqMsoCBDWsTDjBkQ/viewform

ChengshuLi, 2021-09-17:
@DavidYaonanZhu 
Could you please follow our latest documentation and try again?
http://svl.stanford.edu/igibson/docs/dataset.html#download-gibson-and-stanford-2d-3d-semantics-scenes
You should download `2d3ds_for_igibson.zip`. Thanks!

ChengshuLi, 2021-09-22:
Please feel free to re-open the issue if the problem persists. Thanks!

closed 2021-09-22 by ChengshuLi