StanfordVL/iGibson #109: Cannot set RGB observation shape

sparisi, 2021-08-04:
I would like to manually set it from my own script rather than by using a YAML file.
I try

```
        env = iGibsonEnv(config_file=config_file, mode='headless')

        config = parse_config(config_file)
        config['max_step'] = horizon
        config['image_width'] = width
        config['image_height'] = height

        env.config = config
        env.simulator.reload()
        env.load()
```

`max_step` is set correctly, but `image_width` and `image_height` are not. 
Strangely, `env.observation_space.shape['rgb'].shape` changes correctly, but whenever I call `reset()` or `step()` the observation returned is still of the original size.

ChengshuLi, 2021-08-04:
Once `env = iGibsonEnv(config_file=config_file, mode='headless')` is called, the image size will already be set. Is there a particular reason why you don't want to set the image width and height in the YAML file?

sparisi, 2021-08-04:
To have the same structure of other environments. I can do the same with Habitat, DM control, Mujoco, ... So for instance if I need to I can use the same network (same input size) on all environments. It would be more convenient to do it from my python code rather than having to edit the YAML all the time, but it is not crucial. Is there a way to edit the config read by the YAML file before passing its parameter to the simulator?
At the moment I can only pass the path to the YAML to the simulator (not the config itself). 
Thanks for your help!

ChengshuLi, 2021-08-04:
I think you can actually pass the config itself, rather than the config file path. Could you try it out? 

Ref: https://github.com/StanfordVL/iGibson/blob/5f8d253694b23b41c53959774203ba5787578b74/igibson/utils/utils.py#L17-L37

closed 2021-08-04 by sparisi