StanfordVL/iGibson #110: File not found error: data/ig_dataset/metadata/non_sampleable_categories.txt

jadkins99, 2021-08-06:
I followed the initial Gibson setup instructions and installed the dataset using:
 python -m igibson.utils.assets_utils --download_ig_dataset

However, when attempting to import iGibsoEnv with from igibson.envs.igibson_env import iGibsonEnv, I get the following file not found error:
FileNotFoundError: [Errno 2] No such file or directory: '/scratch/jadkins/data/ig_dataset/metadata/non_sampleable_categories.txt

I have also tried downloading and extracting the ig_dataset directly and this file is still missing. How do I obtain non_sampleable_categories.txt?

fxia22, 2021-08-06:
Are you using iGibson 1.0 or 2.0? If you are using iGibson 2.0, please follow the guide here: https://docs.google.com/document/d/1u2m9Ld6Qo3eG-fvCzuAZN6lHxwpVVBlwLVdzo_WDlNI

We will update our documentation to make it clear.

jadkins99, 2021-08-07:
That guide fixed the problem! Thanks!

closed 2021-08-07 by jadkins99